destan19/OpenAppFilter #3: 能出下openwrt编译如何添加你的插件吗

ITdesk01, 2019-10-11:
HI
我尝试编译你的插件，结果失败了，步骤如下
1.我把你的插件git clone到了Openwrt/lean_test/lede/package/Extra-plugin/OpenAppFilter这个目录
2. ./scripts/feeds/update -a  &&  ./scripts/feeds/install -a
3. 我在make menuconfig里面已经看到了，选择*。然后编译，安装到的机器，结果发现luci-app-oaf已经安装，但我在网络里面找不到应用过滤这个选项

我之后尝试了单独编译插件安装，出现以下错误
root@OpenWrt:~# cd /tmp
root@OpenWrt:/tmp# opkg install luci-app-oaf_1.0-1_all.ipk
Installing luci-app-oaf (1.0-1) to root...
Configuring luci-app-oaf.
uci: Entry not found
uci: Parse error (invalid command) at line 1, byte 0

RealKiro, 2019-10-14:
> HI
> 我尝试编译你的插件，结果失败了，步骤如下
> 1.我把你的插件git clone到了Openwrt/lean_test/lede/package/Extra-plugin/OpenAppFilter这个目录
> 2. ./scripts/feeds/update -a && ./scripts/feeds/install -a
> 3. 我在make [...]

[image]
勾选后编译 可以在`网络`找到该插件

ITdesk01, 2019-10-14:
并没有找到此设置，能否提供此插件所需要的依赖

RealKiro, 2019-10-14:
> 并没有找到此设置，能否提供此插件所需要的依赖

缺依赖报错请截图，或者用 https://github.com/project-openwrt/openwrt 按照说明重新更新一下依赖

destan19, 2019-10-16:
最新版本已经加了依赖

ITdesk01, 2019-10-16:
我找时间测试一下

closed 2019-10-19 by ITdesk01