destan19/OpenAppFilter #27: 路由上用官方的固件，单独编译ipk能否可行

liuxiaochakan, 2020-03-08:
[image]
路由上想用官方的固件，只是想单独安装oaf，单独编译后无法安装，有什么办法解决吗？求指点

EverlastingVidfyoi, 2020-03-10:
安装时加上 --force-depends试试

wangwanggogo19xx, 2020-03-13:
请问楼上问题解决了吗？

liuxiaochakan, 2020-03-14:
> 请问楼上问题解决了吗？

[image]
报错还在，但是看起来装上去了，Luci也出来了

liuxiaochakan, 2020-03-14:
> 安装时加上 --force-depends试试

感谢！！！ Luci已经出来了，应该可以用吧

EverlastingVidfyoi, 2020-03-14:
总共四个插件，少一个功能肯定不正常，都要装上去的

liuxiaochakan, 2020-03-24:
> 总共四个插件，少一个功能肯定不正常，都要装上去的

我这边编译出来的少一个中文语言包，而且好像功能还是不正常，一直加载数据中，看来还是不行。应该有办法单独编译ipk，安装在官方固件上吧，希望大牛们指导一下。

EverlastingVidfyoi, 2020-03-24:
你都能编译出来了，要不就编译进固件然后升级固件直接用也行

luuziiyoo, 2020-04-07:
四个ipk都编译好了 也安装好了但是访问记录里还是一直是   正在收集数据。。。。

xswz8015, 2020-04-14:
我在官方中加入一起编译时，因luci版本（git-20.**）不同，界面会出错，无法设置。

destan19, 2020-04-22:
除非保证源码和配置一模一样， 特别是内核的版本和配置

xswz8015, 2020-04-22:
我现在是加入lean的代码编译的，能用。

liuxiaochakan, 2020-04-23:
> 我现在是加入lean的代码编译的，能用。

lean的代码在哪，我看lede和openwrt里面都没有OpenAppFilter啊

luuziiyoo, 2020-04-23:
> > 我现在是加入lean的代码编译的，能用。
> 
> lean的代码在哪，我看lede和openwrt里面都没有OpenAppFilter啊

正因为没有才要单独编译啊有了还用编译啊

luuziiyoo, 2020-04-26:
自己添加进去啊

closed 2022-07-19 by destan19