destan19/OpenAppFilter #42: 请教下 如何在开启SSR科学上网插件的环境下开启应用过滤呢？

jackchim, 2020-04-14:
Q：该插件在ssr-plus 插件开启下不能过滤应用。
请问下大佬，我要实现在开启SSR科学上网插件的环境下开启应用过滤，应该从那些文件函数着手呢？

mengxd, 2020-04-14:
SSR开启后也可以使用网络过滤啊，只要你没开启网络加速（Turbo acc）

jackchim, 2020-04-15:
主机型号	Newifi-D2
架构	MediaTek MT7621 ver:1 eco:3 x 4
固件版本	OpenWrt R9.11.23 / LuCI Master (git-20.088.29068-cebe120)
内核版本	4.14.155

我试过关闭 luci-app-flowoffload 和相关acc加速插件情况下，开启ssr插件 （出口IP是国内）过滤还是不生效。奇怪了。。

destan19, 2020-05-14:
> 主机型号 Newifi-D2
> 架构 MediaTek MT7621 ver:1 eco:3 x 4
> 固件版本 OpenWrt R9.11.23 / LuCI Master (git-20.088.29068-cebe120)
> 内核版本 4.14.155
> 
> 我试过关闭 luci-app-flowoffload 和相关acc加速插件情况下，开启ssr插件 （出口IP是国内）过滤还是不生效。奇怪了。。

可以先用原版的openwrt试试效果

secretandanon, 2021-04-10:
@jackchim 肯定是不行的，因为那些软件都会在prerouting中转发 https://github.com/destan19/OpenAppFilter/issues/102

closed 2022-07-19 by destan19