destan19/OpenAppFilter #45: 官方openwrt的master源码编译该插件到斐讯K3上，会引起间隔性断网，其他均正常！！！

zxlhhyccc, 2020-04-23:
排查了近20天，终于确定官方openwrt的master源码编译该插件到斐讯K3上，会引起间隔性断网，但编译其他固件却正常，编译在18.06和19.07上面也正常。请大佬看看是怎么回事！！！

zwillhill, 2020-05-01:
编译到lean的固件里也发生了类似的情况，插件没启用就会间歇性断网（关了所有网络加速）。没确定是哪个插件的问题。

destan19, 2020-05-04:
这个问题在处理，但不知道怎么复现，你们那边有什么立即出发断网的条件吗

Aliceliese, 2020-05-06:
> 这个问题在处理，但不知道怎么复现，你们那边有什么立即出发断网的条件吗
我是使用的Newifi3 D2路由器，从04.26开始之后我编译的固件只要添加了软件过滤就会触发路由器不断重启，经过5-6个固件的实验发现，无论是否开启过滤，只要固件中有该插件，都会出现在打开网易云后整个路由器在几分钟到十几分钟时间不等内重启，并且必然重现。
根据前几天的测试情况，只要不打开网易云，好像就不会出现重启。
lean大说是因为HWNAT 导致的内核崩溃，但我也试过不编译整个turbo加速，但还是会出现该情况。

05.06 今天在一位我在lean大那边发的issue下评论的朋友的启发下，今天尝试在编译固件时去掉了oaf过滤，然后尝试打开网易云，在使用了一天后发现问题没有复现，路由器不再重启。

结论：由于在04.26以前我也是会编译加入oaf的，但没有出现路由器重启的问题，因此看来应该是lean大那边最近的更新导致了与oaf插件的冲突才导致了路由器重启。

1.顺便给出我在lean大那边发的issue：https://github.com/coolsnowwolf/lede/issues/4531#issue-611581358
包含了log和问题描述；
2.顺便给出在04.27和04.30编译的问题固件，有条件的话可以用于问题复现：https://lyhd.lanzous.com/icc34md

destan19, 2020-05-14:
找到了一个导致内核崩溃的问题，由于连接跟踪状态引起，与新版本内核有一定关系， 已经修复
用最新代码编译测试看是否还存在死机重启问题

closed 2022-07-19 by destan19