destan19/OpenAppFilter #49: 刚更新的最新源码编译报错！！！

zxlhhyccc, 2020-05-24:
[image]

Aes64X, 2020-05-24:
未见异常，或许是指令集问题？（我在IPQ4019 ARMv7上编译，一切正常）

Aes64X, 2020-05-24:
呃。。。。5.4  内核 --

zxlhhyccc, 2020-05-24:
> 呃。。。。5.4 内核 --

commit a4eb6e3674ca3ab963971d2c9fb52e9419f567d3编译正常。。。

Aes64X, 2020-05-24:
> > 呃。。。。5.4 内核 --
> 
> commit a4eb6e3674ca3ab963971d2c9fb52e9419f567d3编译正常。。。

https://github.com/destan19/OpenAppFilter/commit/a7459e61efa12050e5337ec9c8a4e176a024c973

在这里。。。但我提供不了更多帮助了 --

zxlhhyccc, 2020-05-24:
> > > 呃。。。。5.4 内核 --
> > 
> > 
> > commit a4eb6e3674ca3ab963971d2c9fb52e9419f567d3编译正常。。。
> 
> [a7459e6](https://github.com/destan19/OpenAppFilter/commit/a7459e61efa12050e5337ec9c8a4e176a024c973)
> 
> 在这里。。。但我提供不了更多帮助了 --

就是这个commit更新后编译报错。。。。

fuqiang03, 2020-05-24:
4.19内核 x86同样错误

Aes64X, 2020-05-25:
那么。。。只有4.14内核是正常的？我这还是没报错。。。。。

garypang13, 2020-05-25:
同样报错,5.14内核

2020-05-25T01:59:24.2296904Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/app_filter.c: In function 'init_oaf_timer':
2020-05-25T01:59:24.2297834Z /home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/app_filter.c:915:5: error: implicit declaration of function 'setup_timer'; did you mean 'sk_stop_timer'? [-Werror=implicit-function-declaration]
2020-05-25T01:59:24.2298045Z      setup_timer(&oaf_timer, oaf_timer_func, OAF_TIMER_INTERVAL * HZ);
2020-05-25T01:59:24.2298191Z      ^~~~~~~~~~~
2020-05-25T01:59:24.2298317Z      sk_stop_timer
2020-05-25T01:59:24.4872131Z cc1: some warnings being treated as errors
2020-05-25T01:59:24.4929144Z scripts/Makefile.build:265: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/app_filter.o' failed
2020-05-25T01:59:24.4930183Z make[5]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/app_filter.o] Error 1
2020-05-25T01:59:24.4932984Z make[4]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
2020-05-25T01:59:24.4933555Z Makefile:1700: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
2020-05-25T01:59:24.4934441Z make[3]: *** [/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
2020-05-25T01:59:24.4935178Z make[4]: Leaving directory '/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.42'
2020-05-25T01:59:24.4936267Z Makefile:41: recipe for target '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
2020-05-25T01:59:24.4936832Z make[3]: Leaving directory '/home/runner/work/Actions-OpenWrt-Nginx/Actions-OpenWrt-Nginx/openwrt/feeds/custom/luci/oaf'
2020-05-25T01:59:24.4937857Z time: package/feeds/custom/oaf/compile#1.47#0.37#1.85
2020-05-25T01:59:24.4944503Z package/Makefile:111: recipe for target 'package/feeds/custom/oaf/compile' failed

HoldOnBro, 2020-05-25:
lede x86 报错前来集合

Aes64X, 2020-05-25:
目前状况：
4.14 √（竞斗云测试通过，拦截正常，一切正常）
（顺便。。。因为刚需这玩意，所有的内网交换加速模块，SFE之类的都被我直接砍了，各位可以尝试下这种操作，反正你用过滤就没加速，不如砍干净。）
4.19 X
5.14 X

ChrisLiu0343, 2020-05-26:
同样报错大雕的源码~~~~~~~~~~~~~~~~~~~~4.19

destan19, 2020-05-27:
是内核定时器api兼容问题， 4.15以上版本会编译出错，已经修复

yaott2020-666, 2020-07-31:
支持Pandorabox的3.14.79内核版本不？

closed 2022-07-19 by destan19