destan19/OpenAppFilter #57: /usr/lib/lua/luci/model/cbi/appfilter/appfilter.lua:111: attempt to index local 'device' (a nil value)

quintus-lab, 2020-06-18:
最近更新的版本在openwrt 19.07下出错
/usr/lib/lua/luci/model/cbi/appfilter/appfilter.lua:111: attempt to index local 'device' (a nil value)
stack traceback:
	/usr/lib/lua/luci/model/cbi/appfilter/appfilter.lua:111: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1335: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1018: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:478: in function </usr/lib/lua/luci/dispatcher.lua:477>

destan19, 2020-06-18:
可能是兼容性问题， 18.06版本没问题

ledewrt, 2020-06-19:
我这里是1907系列没有问题，1806系列有问题：
/usr/lib/lua/luci/dispatcher.lua:381: Failed to execute cbi dispatcher target for entry '/admin/control/appfilter'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/appfilter/appfilter.lua:89: attempt to index local 't' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:381: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:95: in function </usr/lib/lua/luci/dispatcher.lua:94>

closed 2020-07-11 by destan19