destan19/OpenAppFilter #61: pandorabox编译失败

yaott2020-666, 2020-07-30:
使用Pandorabox的SDK编译出错，路由器是新三 mt7621

echo "kmod-ralink-hwnat" >> /root/pan/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/linux.default.install
if [ -f /root/pan/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/linux.default.install.clean ]; then rm -f /root/pan/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/linux.default.install /root/pan/staging_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/pkginfo/linux.default.install.clean; fi
make[2]: Leaving directory '/root/pan/package/linux'
make[2]: Entering directory '/root/pan/package/OpenAppFilter/oaf'
make -C "/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/linux-3.14.79"  M="/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/oaf-3.0" EXTRA_CFLAGS=""  modules
make[3]: Entering directory '/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/linux-3.14.79'
Makefile:590: /root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/linux-3.14.79/arch/x86/Makefile: No such file or directory
make[3]: *** No rule to make target '/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/linux-3.14.79/arch/x86/Makefile'.  Stop.
make[3]: Leaving directory '/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/linux-3.14.79'
Makefile:41: recipe for target '/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/oaf-3.0/.built' failed
make[2]: *** [/root/pan/build_dir/target-mipsel_1004kc+dsp_uClibc-1.0.x/linux-ralink_mt7621/oaf-3.0/.built] Error 2
make[2]: Leaving directory '/root/pan/package/OpenAppFilter/oaf'
package/Makefile:105: recipe for target 'package/OpenAppFilter/oaf/compile' failed
make[1]: *** [package/OpenAppFilter/oaf/compile] Error 2
make[1]: Leaving directory '/root/pan'
/root/pan/include/toplevel.mk:192: recipe for target 'package/oaf/compile' failed
make: *** [package/oaf/compile] Error 2

无法编译出kmod-oaf

syb999, 2020-08-05:
The reason for the above error is that the kernel is less than 5. X：
U can use "SUBDIRS=" instead of "M=".

Edit /oaf/Makefile:
-	M="$(PKG_BUILD_DIR)" \
+	SUBDIRS="$(PKG_BUILD_DIR)" \

yaott2020-666, 2020-08-06:
还是不行，一样的报错，应该是内核版本太低了。

syb999, 2020-08-07:
3.14内核应该没问题的。
我这里3.10内核，3.18内核都能正常使用。

yaott2020-666, 2020-08-08:
可以具体说一下用Pandorabox的sdk怎么编译的吗，我用的是mt7621的sdk

destan19, 2021-02-21:
这个需要自行适配

closed 2021-02-21 by destan19