destan19/OpenAppFilter #63: disable sfe脚本是有问题的

wlshdjj, 2020-09-07:
if [ x"$enable" != x"1" ];then
		echo "appfilter is disabled"
		echo 0 >/proc/sys/oaf/enable>/dev/null
		uci set firewall.@defaults[0].flow_offloading=1
		return 0
    else
		insmod oaf >/dev/null
		echo 1 >/proc/sys/oaf/enable
		uci set firewall.@defaults[0].flow_offloading=0
	fi
首先，uci set只是在内存中修改，需要uci commit才是真正修改。修改之后还要reload一次才会使用新的配置。
其次，这代码并没有检测人家原先是否启用，人家本来没有启用的，你关闭之后反而把人家的启用了，可能无意中造成了新的问题了。

destan19, 2020-09-19:
加这几行代码关闭sfe确实有点别扭，之前主要考虑很多人反馈加速了不能用，但又不知道原因
后面还是去掉吧

Archiewyq, 2020-11-09:
> 加这几行代码关闭sfe确实有点别扭，之前主要考虑很多人反馈加速了不能用，但又不知道原因
> 后面还是去掉吧

开了sfe或者qca-nss-ecm后，你在FORWARD上的hook就失效了(sfe的匹配在FORWARD之前，匹配到之后直接调dev_queue_xmit发送了)，我的解决办法是在sfe里再加一个hook专门处理经过sfe加速的流量。

destan19, 2021-01-26:
sfe自行关闭，暂不支持同时开turbo acc和应用过滤

closed 2021-01-26 by destan19