destan19/OpenAppFilter #81: oaf在官方master的最新内核5.4.91下编译报错！

zxlhhyccc, 2021-01-20:
[image]

sxx1314, 2021-01-20:
确实报错了。上游改动导致的问题。临时解决办法是直接屏蔽相应函数但是使用会出问题。

zxlhhyccc, 2021-01-21:
> 确实报错了。上游改动导致的问题。临时解决办法是直接屏蔽相应函数但是log等会出问题，自测倒是不影响使用，还是 @destan19 来修复修复。
> 
> ```
> t/5.4kernel/staging_dir/hostpkg/include -I/home/openwrt/openwrt/5.4kernel/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Jan 20 10:55:01 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/openwrt/openwrt/5.4kernel/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/openwrt/openwrt/5.4kernel/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=5.4.91 M="/home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0" EXTRA_CFLAGS=""  modules
> make[4]: Entering directory '/home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
>   CC [M]  /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o
> /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_init_log_sysctl':
> /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:56:27: warning: unused variable 'hdr' [-Wunused-variable]
>   struct ctl_table_header *hdr;
>                            ^~~
> /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_init':
> /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:75:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_exit':
> /home/openwrt/openwrt/5.4kernel/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:79:1: error: control reaches end of non-void function [-Werror=return-type]
>  }
>  ^
> cc1: some warnings being treated as errors
> ```

请问屏蔽哪些函数？

ma20at, 2021-01-22:
> 请问屏蔽哪些函数？

报错日志不是写着吗？屏蔽af_log.c源码里的af_log_init，af_log_exit这2个function就可以编译了

lxwq, 2021-01-22:
编译通不过,坐等修复
  CC [M]  /home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o
/home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_init_log_sysctl':
/home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:56:27: warning: unused variable 'hdr' [-Wunused-variable]
  struct ctl_table_header *hdr;
                           ^~~
/home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_init':
/home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:75:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_exit':
/home/zz/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:79:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^

xylz0928, 2021-01-22:
有相同问题

80houco, 2021-01-22:
是的 失败。。。。。

80houco, 2021-01-22:
cc1: some warnings being treated as errors
scripts/Makefile.build:261: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o' failed
make[5]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o] Error 1
Makefile:1732: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
make[4]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
make[4]: Leaving directory '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
Makefile:41: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
make[3]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
make[3]: Leaving directory '/home/lh/lede/package/lean/OpenAppFilter/oaf'
time: package/lean/OpenAppFilter/oaf/compile#2.62#0.43#3.06
package/Makefile:111: recipe for target 'package/lean/OpenAppFilter/oaf/compile' failed
make[2]: *** [package/lean/OpenAppFilter/oaf/compile] Error 2
make[2]: Leaving directory '/home/lh/lede'
package/Makefile:107: recipe for target '/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lh/lede'
/home/lh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

ChrisLiu0343, 2021-01-22:
相同问题。。。

huaduxu, 2021-01-23:
按下面写法可以通过`int af_log_init(void){
	af_init_log_sysctl();
	return 0;
}

int af_log_exit(void){
	af_fini_log_sysctl();
	return 0;`

cqranjian, 2021-01-23:
> 按下面写法可以通过`int af_log_init(void){
> af_init_log_sysctl();
> return 0;
> }
> 
> int af_log_exit(void){
> af_fini_log_sysctl();
> return 0;`

影响使用么

ma20at, 2021-01-23:
> > 按下面写法可以通过`int af_log_init(void){
> > af_init_log_sysctl();
> > return 0;
> > }
> > int af_log_exit(void){
> > af_fini_log_sysctl();
> > return 0;`
> 
> 影响使用么

好像正常运行，不过这样跳过之后启动系统oaf模块加载不上，
  19.004641] oaf: Unknown symbol af_log_exit (err -2)
[   19.005563] oaf: Unknown symbol af_log_init (err -2)
你可以把正在运行的旧版本oaf.ko覆盖的新系统的/lib/modules/5.4.91/oaf.ko后就可以正常加载

destan19, 2021-01-26:
增加函数返回值， 已经修复

closed 2021-01-26 by destan19