destan19/OpenAppFilter #82: 编译失败

80houco, 2021-01-22:
项目 https://github.com/coolsnowwolf/lede ，编译失败
cc1: some warnings being treated as errors
scripts/Makefile.build:261: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o' failed
make[5]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o] Error 1
Makefile:1732: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
make[4]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
make[4]: Leaving directory '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
Makefile:41: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
make[3]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
make[3]: Leaving directory '/home/lh/lede/package/lean/OpenAppFilter/oaf'
time: package/lean/OpenAppFilter/oaf/compile#2.62#0.43#3.06
package/Makefile:111: recipe for target 'package/lean/OpenAppFilter/oaf/compile' failed
make[2]: *** [package/lean/OpenAppFilter/oaf/compile] Error 2
make[2]: Leaving directory '/home/lh/lede'
package/Makefile:107: recipe for target '/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lh/lede'
/home/lh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

CWZ7605, 2021-01-22:
项目project-openwrt编译失败
` /home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/af_log.c: In function 'af_log_init':
/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/af_log.c:75:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/af_log.c: In function 'af_log_exit':
/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/af_log.c:79:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[5]: *** [/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/af_log.o] Error 1
scripts/Makefile.build:261: recipe for target '/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/af_log.o' failed
make[4]: *** [/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0] Error 2
Makefile:1732: recipe for target '/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0' failed
make[3]: *** [/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/.built] Error 2
make[4]: Leaving directory '/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.4.91'
Makefile:51: recipe for target '/home/runner/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-3.0/.built' failed`

cqranjian, 2021-01-23:
make[4]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
  CC [M]  /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_init_log_sysctl':
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:56:27: warning: unused variable 'hdr' [-Wunused-variable]
  struct ctl_table_header *hdr;
                           ^~~
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_init':
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:75:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_exit':
/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:79:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
make[5]: *** [/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o] Error 1
scripts/Makefile.build:261: recipe for target '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o' failed
make[4]: *** [/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
Makefile:1732: recipe for target '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
make[3]: *** [/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
make[4]: Leaving directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
Makefile:41: recipe for target '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
make[3]: Leaving directory '/home/runner/work/lede/lede/package/openwrt-packages/OpenAppFilter/oaf'
time: package/openwrt-packages/OpenAppFilter/oaf/compile#0.96#0.32#2.12
package/Makefile:111: recipe for target 'package/openwrt-packages/OpenAppFilter/oaf/compile' failed
make[2]: *** [package/openwrt-packages/OpenAppFilter/oaf/compile] Error 2
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/home/runner/work/lede/lede'
/home/runner/work/lede/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

hong0980, 2021-01-23:
openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:79:1: error: control reaches end of non-void function [-Werror=return-type]

closed 2021-02-21 by destan19