destan19/OpenAppFilter #84: 编译失败

jqglv, 2021-01-24:
用Lean的Openwrt源码编译，每次都出错

A：cc1: some warnings being treated as errors

scripts/Makefile.build:261: recipe for target '/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o' failed
make[5]: *** [/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o] Error 1
Makefile:1732: recipe for target '/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
make[4]: *** [/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
make[4]: Leaving directory '/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
Makefile:41: recipe for target '/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
make[3]: *** [/home/mc/openwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
make[3]: Leaving directory '/home/mc/openwrt/lede/package/OpenAppFilter/oaf'
time: package/OpenAppFilter/oaf/compile#0.69#0.30#0.96
package/Makefile:111: recipe for target 'package/OpenAppFilter/oaf/compile' failed
make[2]: *** [package/OpenAppFilter/oaf/compile] Error 2
make[2]: Leaving directory '/home/mc/openwrt/lede'
package/Makefile:107: recipe for target '/home/mc/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/mc/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mc/openwrt/lede'
/home/mc/openwrt/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Miroxyz, 2021-01-24:
我也有一样的问题

hououinkami, 2021-01-26:
+1最近才出现

keeno1982, 2021-01-26:
已经修复了吧，我刚编译通过

zxlhhyccc, 2021-01-26:
不是log问题而是app_filter.c编译报错：
~~~
oaf-3.0" EXTRA_CFLAGS=""  modules
make[3]: Entering directory '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.92'
  CC [M]  /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.o
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function '__add_app_feature':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:110:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  110 |   char *p = dict;
      |   ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'add_app_feature':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:158:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  158 |  char *p = feature;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:202:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  202 |  int dst_port = 0;
      |  ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'af_init_feature':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:232:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  232 |  char *p = feature_str;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:247:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  247 |  char feature[MAX_FEATURE_STR_LEN];;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:248:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  248 |  int i;
      |  ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:251:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  251 |  char *begin = feature_buf;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:248:6: warning: unused variable 'i' [-Wunused-variable]
  248 |  int i;
      |      ^
In file included from /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:24:
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'load_feature_buf_from_file':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:284:11: warning: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'off_t' {aka 'long int'} [-Wformat=]
  284 |  AF_DEBUG("feature file size: %u\n", size);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~
      |                                      |
      |                                      off_t {aka long int}
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/af_log.h:7:16: note: in definition of macro 'LOG'
    7 |         printk(fmt, ##__VA_ARGS__); \
      |                ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:284:2: note: in expansion of macro 'AF_DEBUG'
  284 |  AF_DEBUG("feature file size: %u\n", size);
      |  ^~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:284:32: note: format string is defined here
  284 |  AF_DEBUG("feature file size: %u\n", size);
      |                               ~^
      |                                |
      |                                unsigned int
      |                               %lu
In file included from /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:24:
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:286:11: warning: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'off_t' {aka 'long int'} [-Wformat=]
  286 |   AF_WARN("warning, file size = %u\n", size);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~
      |                                        |
      |                                        off_t {aka long int}
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/af_log.h:7:16: note: in definition of macro 'LOG'
    7 |         printk(fmt, ##__VA_ARGS__); \
      |                ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:286:3: note: in expansion of macro 'AF_WARN'
  286 |   AF_WARN("warning, file size = %u\n", size);
      |   ^~~~~~~
[... cut ...]
time: package/bf-package-master/zxlhhyccc/OpenAppFilter/oaf/compile#2.81#0.92#3.72
    ERROR: package/bf-package-master/zxlhhyccc/OpenAppFilter/oaf failed to build.
package/Makefile:112: recipe for target 'package/bf-package-master/zxlhhyccc/OpenAppFilter/oaf/compile' failed
make[1]: *** [package/bf-package-master/zxlhhyccc/OpenAppFilter/oaf/compile] Error 1
make[1]: Leaving directory '/home/lin/openwrt-luci-master'
/home/lin/openwrt-luci-master/include/toplevel.mk:232: recipe for target 'package/oaf/compile' failed
make: *** [package/oaf/compile] Error 2
~~~

HoldOnBro, 2021-01-27:
x86 架构 target 设置为gemini lake 同样编译失败

destan19, 2021-02-21:
已经修复了相关警告

closed 2021-02-21 by destan19