destan19/OpenAppFilter #87: https://github.com/coolsnowwolf/lede；项目编译出错

80houco, 2021-01-24:
CC [M]  /home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o
/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_init_log_sysctl':
/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:56:27: warning: unused variable 'hdr' [-Wunused-variable]
  struct ctl_table_header *hdr;
                           ^~~
/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_init':
/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:75:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c: In function 'af_log_exit':
/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.c:79:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: some warnings being treated as errors
scripts/Makefile.build:261: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o' failed
make[5]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/af_log.o] Error 1
Makefile:1732: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0' failed
make[4]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0] Error 2
make[4]: Leaving directory '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.91'
Makefile:41: recipe for target '/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built' failed
make[3]: *** [/home/lh/lede/build_dir/target-x86_64_musl/linux-x86_64/oaf-3.0/.built] Error 2
make[3]: Leaving directory '/home/lh/lede/package/OpenAppFilter/oaf'
time: package/OpenAppFilter/oaf/compile#2.71#0.38#3.31
package/Makefile:111: recipe for target 'package/OpenAppFilter/oaf/compile' failed
make[2]: *** [package/OpenAppFilter/oaf/compile] Error 2
make[2]: Leaving directory '/home/lh/lede'
package/Makefile:107: recipe for target '/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/lh/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lh/lede'
/home/lh/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

db-one, 2021-01-25:
同样报错

tianiue, 2021-01-25:
5.4.91内核报错，4.19内核正常

richaradlee, 2021-01-25:
同样的问题，报错~~~

destan19, 2021-01-26:
增加函数返回值，已经修复

closed 2021-01-26 by destan19