destan19/OpenAppFilter #88: oaf更新了部分代码，但还是在官方master的内核5.4.91及以上下编译失败！

zxlhhyccc, 2021-01-26:
·········
oaf-3.0" EXTRA_CFLAGS=""  modules
make[3]: Entering directory '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.92'
  CC [M]  /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.o
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function '__add_app_feature':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:110:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  110 |   char *p = dict;
      |   ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'add_app_feature':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:158:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  158 |  char *p = feature;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:202:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  202 |  int dst_port = 0;
      |  ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'af_init_feature':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:232:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  232 |  char *p = feature_str;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:247:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  247 |  char feature[MAX_FEATURE_STR_LEN];;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:248:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  248 |  int i;
      |  ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:251:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  251 |  char *begin = feature_buf;
      |  ^~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:248:6: warning: unused variable 'i' [-Wunused-variable]
  248 |  int i;
      |      ^
In file included from /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:24:
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'load_feature_buf_from_file':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:284:11: warning: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'off_t' {aka 'long int'} [-Wformat=]
  284 |  AF_DEBUG("feature file size: %u\n", size);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~
      |                                      |
      |                                      off_t {aka long int}
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/af_log.h:7:16: note: in definition of macro 'LOG'
    7 |         printk(fmt, ##__VA_ARGS__); \
      |                ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:284:2: note: in expansion of macro 'AF_DEBUG'
  284 |  AF_DEBUG("feature file size: %u\n", size);
      |  ^~~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:284:32: note: format string is defined here
  284 |  AF_DEBUG("feature file size: %u\n", size);
      |                               ~^
      |                                |
      |                                unsigned int
      |                               %lu
In file included from /home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:24:
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:286:11: warning: format '%u' expects argument of type 'unsigned int', but argument 2 has type 'off_t' {aka 'long int'} [-Wformat=]
  286 |   AF_WARN("warning, file size = %u\n", size);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~
      |                                        |
      |                                        off_t {aka long int}
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/af_log.h:7:16: note: in definition of macro 'LOG'
    7 |         printk(fmt, ##__VA_ARGS__); \
      |                ^~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:286:3: note: in expansion of macro 'AF_WARN'
  286 |   AF_WARN("warning, file size = %u\n", size);
      |   ^~~~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:286:34: note: format string is defined here
  286 |   AF_WARN("warning, file size = %u\n", size);
      |                                 ~^
      |                                  |
      |                                  unsigned int
      |                                 %lu
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:293:10: error: 'return' with a value, in function returning void [-Werror=return-type]
  293 |   return -1;
[... 67 lines cut ...]
      |                                     |
      |                                     int
      |                                    %ld
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:742:6: warning: unused variable 'found' [-Wunused-variable]
  742 |  int found = 0;
      |      ^~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:735:6: warning: unused variable 'i' [-Wunused-variable]
  735 |  int i;
      |      ^
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'af_update_client_app_info':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:769:6: warning: unused variable 'index' [-Wunused-variable]
  769 |  int index = 0;
      |      ^~~~~
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:768:6: warning: unused variable 'i' [-Wunused-variable]
  768 |  int i;
      |      ^
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c: In function 'app_filter_hook':
/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.c:794:21: warning: unused variable 'total_packets' [-Wunused-variable]
  794 |  unsigned long long total_packets = 0;
      |                     ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:261: recipe for target '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.o' failed
make[4]: *** [/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0/app_filter.o] Error 1
Makefile:1732: recipe for target '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/oaf-3.0' failed
······

destan19, 2021-01-28:
是编译选项更严格了，警告当成了错误，我抽时间看看

zxlhhyccc, 2021-01-28:
> 是编译选项更严格了，警告当成了错误，我抽时间看看

谢谢dalao！！！

closed 2021-02-21 by destan19