destan19/OpenAppFilter #89: luci-app-oaf在官方master分支下页面出错！

zxlhhyccc, 2021-01-30:
[image]

zxlhhyccc, 2021-01-30:
经查，发现是更新了此段代码引起，删除正常。。。
[image]

zxlhhyccc, 2021-01-30:
临时办法：
将if config_users~=nil then改为：if config_users==nil then即可，请dalao修复，谢谢！

lxwq, 2021-03-23:
[image]
和你上面的报错信息不一样的,我的好像是找不到文件,
在编译时也有错误信息,有几个文件找不到,

lxwq, 2021-03-24:
[image]
编译时的错误信息,

lxwq, 2021-03-26:
> [image]
> 编译时的错误信息,

找到原因了,原来是因为luci-compat没有安装,编译上以后可以打开了,
但都是英文的,appfilter的主菜单也是英文,不知道还少什么东西没装,

lxwq, 2021-03-26:
[image]
菜单是英文的,好像PO文件没起做用,这是怎么回事啊?
加了几个其它的软件包试了试,也是一样,
只有官方就在的软件包可以正常显示中文,

d9394, 2021-04-05:
> 
> 
> 经查，发现是更新了此段代码引起，删除正常。。。
> [image]

请问具体是修改哪个文件？

closed 2022-07-19 by destan19