destan19/OpenAppFilter #90: 过滤不生效,正在收集数据...

lxwq, 2021-02-02:
[20210201.zip](https://github.com/destan19/OpenAppFilter/files/5909114/20210201.zip)
LEAN源码的配置文件,哪位大哥有空的话试试看,指正一下哪的问题,
[image]
编译出来后就是这个样子,正在收集数据...
oaf是加载的,

root@OpenWrt:~# lsmod |grep oaf
oaf 49152 0

zxlhhyccc, 2021-02-02:
> [20210201.zip](https://github.com/destan19/OpenAppFilter/files/5909114/20210201.zip)
> LEAN源码的配置文件,哪位大哥有空的话试试看,指正一下哪的问题,
> [image]
> 编译出来后就是这个样子,正在收集数据...
> oaf是加载的,
> 
> root@OpenWrt:~# lsmod |grep oaf
> oaf 49152 0

官方master分支下没问题。。。

wlshdjj, 2021-03-03:
luci 1.x的不能使用，大家散了吧。

closed 2021-03-11 by lxwq