destan19/OpenAppFilter #96: commit 4483aab7d158031adc99b8669d5e7cf4f1023c14版本出了大乱子！！！

zxlhhyccc, 2021-03-04:
[image]

wlshdjj, 2021-03-05:
就你问题多，哈！

zxlhhyccc, 2021-03-05:
> 就你问题多，哈！

提bug可以更好的改进。。。

wlshdjj, 2021-03-05:
我有强大得多的更新脚本，可以给出新增和删减的条目。但我不会用github。

wlshdjj, 2021-03-05:
调试的结果：

wlshdjj, 2021-03-05:
Fri Mar  5 12:42:44 2021 user.notice root: Appfilter's feature.cfg 已成功更新，原来规则 179 条，现在规则 179 条。
Fri Mar  5 12:42:44 2021 user.notice root: Appfilter 有用户自定义规则1条。
Fri Mar  5 12:42:44 2021 user.notice root: Appfilter --------------------删减规则如下：--------------------
Fri Mar  5 12:42:44 2021 user.notice root: 1011 frerdddd:[tcp;;;ppkankan;;]
Fri Mar  5 12:42:44 2021 user.notice root: Appfilter --------------------规则删减完成！--------------------
Fri Mar  5 12:42:44 2021 user.notice root: Appfilter --------------------新增规则如下：--------------------
Fri Mar  5 12:42:44 2021 user.notice root: 1001 QQ:[tcp;;;;;00:02|-1:03,tcp;;;;;02:02|-1:03,tcp;;14000;;;,tcp;;8080;;;00:ca|01:3c,tcp;;;;;00:00|01:00|02:00|03:15]
Fri Mar  5 12:42:44 2021 user.notice root: Appfilter --------------------规则新增完成！--------------------

zxlhhyccc, 2021-03-05:
> Fri Mar 5 12:42:44 2021 user.notice root: Appfilter's feature.cfg 已成功更新，原来规则 179 条，现在规则 179 条。
> Fri Mar 5 12:42:44 2021 user.notice root: Appfilter 有用户自定义规则1条。
> Fri Mar 5 12:42:44 2021 user.notice [...]

你这个是在哪个分支下安装的？我在官方master分支下编译的出现上面的问题，经查找原因，发现是截图上的两行代码引起的，我修改后正常：（将cbi/other_upload和cbi/other_dvalue改成：cbi/upload和cbi/dvalue正常，因为官方源码里只有upload.htm和dvalue.htm）
[image]
另外一个问题就是，新版无法展示“选择用户”和过滤记录，导致无法对某个客户端进行过滤，见截图：
[image]

zxlhhyccc, 2021-03-05:
> 我有强大得多的更新脚本，可以给出新增和删减的条目。但我不会用github。

能否提供你的更新脚本？我来测试一下可以嘛？

wlshdjj, 2021-03-05:
18.06.8。我不用oaf新版，完全无法使用。我继续老版本去。

zxlhhyccc, 2021-03-05:
旧版展示的选择用户和访问记录：
[image]

zxlhhyccc, 2021-03-05:
> 18.06.8。我不用oaf新版，完全无法使用。我继续老版本去。

能否把你的更新脚本发我测试一下呢？

wlshdjj, 2021-03-05:
`#!/bin/sh
# Auth0r=wulishui , < wulishui@gmail.com >

NX=$(grep -n 'class user' /etc/appfilter/feature.cfg | awk -F ':' '{print $1-1}') 
if [ -n "$NX" ]; then
 cat /etc/appfilter/feature.cfg | head -"$NX" > /tmp/log/feature.cfg.old 
 cat /etc/appfilter/feature.cfg | awk 'BEGIN {RS = "\n\n+";ORS = "\n\n"} /class user/ {print $0}' | awk NF > /etc/appfilter/feature.cfg.user
else
 cp -f /etc/appfilter/feature.cfg /tmp/log/feature.cfg.old
fi

rm -f /tmp/log/afrupdatestatus 2>/dev/null
times_tried=0
while [ $times_tried -le 5 ]
do
curl https://raw.githubusercontent.com/destan19/OpenAppFilter/master/open-app-filter/files/feature.cfg > /tmp/log/feature.cfg.new
[ -s /tmp/log/feature.cfg.new ] && break
times_tried=$((times_tried+1))
done
[ -s /tmp/log/feature.cfg.new ] || { echo "1" > /tmp/log/afrupdatestatus ; logger "Appfilter 更新文件下载失败，更新失败。可以终端输入： sh -vx /usr/bin/appfilterupdate 运行一次查看失败原因。" ; exit 0 ; }

new_md5=`md5sum /tmp/log/feature.cfg.new | awk -F ' ' '{print $1}'`
old_md5=`md5sum /tmp/log/feature.cfg.old | awk -F ' ' '{print $1}'`
[ "$new_md5" = "$old_md5" ] && { echo "2" > /tmp/log/afrupdatestatus ; logger "Appfilter 本地特征库是最新的，无须更新。" ; exit 0 ; }

size_new=`cat /tmp/log/feature.cfg.new|grep -v '#'|awk NF|wc -l`
size_old=`cat /tmp/log/feature.cfg.old|grep -v '#'|awk NF|wc -l`
[ "$size_new" -lt $((size_old*8/10)) ] && { echo "3" > /tmp/log/afrupdatestatus ; logger "Appfilter 更新文件内容少太多(原有 $size_old 条，更新文件 $size_new 条)，有可能是下载中断，更新失败。可到https://raw.githubusercontent.com/destan19/OpenAppFilter/master/open-app-filter/files/feature.cfg手动复制使用。" ; exit 0 ; }
echo "4" > /tmp/log/afrupdatestatus
logger "Appfilter's feature.cfg 已成功更新，原来规则 $size_old 条，现在规则 $size_new 条。"

if [ -s /etc/appfilter/feature.cfg.user ]; then
 sum=$(cat /etc/appfilter/feature.cfg.user|grep -v '#'|wc -l)
 logger "Appfilter 有用户自定义规则"$sum"条。"
 cat /tmp/log/feature.cfg.new /etc/appfilter/feature.cfg.user > /etc/appfilter/feature.cfg
else
 logger "Appfilter 没有用户自定义规则。"
 cat /tmp/log/feature.cfg.new > /etc/appfilter/feature.cfg
fi

deleted=$(grep -Fvxf /tmp/log/feature.cfg.new /tmp/log/feature.cfg.old | grep -v '#')
if [ -n "$deleted" ]; then
logger "Appfilter --------------------删减规则如下：--------------------"
echo "$deleted"|while read line
do
 logger "$line"
done
logger "Appfilter --------------------规则删减完成！--------------------"
fi

newadded=$(grep -Fvxf /tmp/log/feature.cfg.old /tmp/log/feature.cfg.new | grep -v '#')
if [ -n "$newadded" ]; then
logger "Appfilter --------------------新增规则如下：--------------------"
echo "$newadded"|while read line
do
 logger "$line"
done
logger "Appfilter --------------------规则新增完成！--------------------"
fi

rm -rf /tmp/appfilter 2>/dev/null
rm -f /tmp/log/feature.cfg.*
/etc/init.d/appfilter restart

`
我的是支持自定义规则的，自定义规则更新时不会丢失，自定义规则放在最后：
`#class user
9001 这是示例:[tcp;;;www.example.com.jp;;]
`

wlshdjj, 2021-03-05:
你看看将https://raw.githubusercontent.com/destan19/OpenAppFilter/master/open-app-filter/files/feature.cfg换成他的io，因为raw的DNS被污染，需要自己指定host才能更新。

zxlhhyccc, 2021-03-05:
> https://raw.githubusercontent.com/destan19/OpenAppFilter/master/open-app-filter/files/feature.cfg

能否直接上传脚本？你直接贴出来有点乱？谢谢！

destan19, 2021-03-05:
> 你这个是在哪个分支下安装的？我在官方master分支下编译的出现上面的问题，经查找原因，发现是截图上的两行代码引起的，我修改后正常：（将cbi/other_upload和cbi/other_dvalue改成：cbi/upload和cbi/dvalue正常，因为官方源码里只有upload.htm和dvalue.htm）
> [image] [...]

我开发基于lean lede的，openwrt master会存在兼容问题， 主干正在开发阶段，还需要大量兼容性测试，可以先用稳定分支v3.0.1

destan19, 2021-03-05:
[image]
[image]
[image]
开发基于lean 的lede源码，openwrt主干后续适配， 稳定版本为v3.0.1分支，主干会不断增加新功能，难免会存在bug

zxlhhyccc, 2021-03-05:
> [image]
> [image]
> [image]
> 开发基于lean 的lede源码，openwrt主干后续适配， 稳定版本为v3.0.1分支，主干会不断增加新功能，难免会存在bug

dalao，openwrt主干的请尽快适配啊，太漂亮了。。。。

closed 2021-03-06 by destan19