destan19/OpenAppFilter #98: iptables-mod-fullconenat也要关掉吗

lxwq, 2021-03-07:
[image]
但这个组件不知道要怎么取消,和别的有依赖关系,

wlshdjj, 2021-03-08:
不用取消，你只要不开启就行

lxwq, 2021-03-09:
> 不用取消，你只要不开启就行

搞了很久,终于明白是firewall防火墙依赖它,可是要关掉防火墙工具吗?
或者是我搞错了,

lxwq, 2021-03-09:
现在编译出来,运行是这个样子,
[image]
if string.match(fullcone, "1")  or string.match(bbr, "1") or string.match(flow_offloading, "1") then
    um.value="运行环境检测失败，请先关闭ACC加速模块!"
这里面bbr和flow_offloading都没有编译上,只有kmod-ipt-fullconenat和iptables-mod-fullconenat因为防火墙依赖取消不掉,

lxwq, 2021-03-11:
找不到哪的问题,换了lienol的源,防火墙选上编译,运行环境检测正常,

closed 2021-03-11 by lxwq

chentengming, 2021-03-22:
@lxwq 请问你换了lienol的源，是哪个源？

chentengming, 2021-03-22:
lienol的源也有这个问题吧 @lxwq 
local fullcone=SYS.exec("uci get firewall.@defaults[0].fullcone");