destan19/OpenAppFilter #99: 有人试过使用LEAN的源码编译吗

lxwq, 2021-03-13:
在使用LEAN的LEDE源码编译后,总是出现运行环境检测失败,请先关闭ACC加速模块,CLONE最新源码,关了无数软件还是不行,
https://github.com/coolsnowwolf/lede/

keeno1982, 2021-03-16:
是不行，不知道从哪里关

RodmanWang, 2021-03-16:
这个提示你不管就行啊，不影响，实在看不惯，就去网络-防火墙-"启用FullCone-NAT",把这个勾勾去掉就行

lxwq, 2021-03-17:
> 这个提示你不管就行啊，不影响，实在看不惯，就去网络-防火墙-"启用FullCone-NAT",把这个勾勾去掉就行

多谢指点,去掉勾勾后就没有提示了,
另外,
使用官方OP编译时,
使用openwrt-18.06分枝时,没有EFI选项,别的分枝OAF无法正常运行,
有什么办法吗?

ma20at, 2021-03-20:
我用的就是lean的，防火墙fullcone-nat管不管闭这个oaf过滤都不起作用，不知道为什么，内核版本5.4.105

lxwq, 2021-03-20:
> 我用的就是lean的，防火墙fullcone-nat管不管闭这个oaf过滤都不起作用，不知道为什么，内核版本5.4.105

广告屏蔽大师 Plus+里的拦截 短视频 APP 和网站是起做用的,不知道为什么,可能是过滤方式和OAF不一样,

ma20at, 2021-03-20:
> > 我用的就是lean的，防火墙fullcone-nat管不管闭这个oaf过滤都不起作用，不知道为什么，内核版本5.4.105
> 
> 广告屏蔽大师 Plus+里的拦截 短视频 APP 和网站是起做用的,不知道为什么,可能是过滤方式和OAF不一样,

我知道怎么弄了，kp里访问控制把要屏蔽的设备mac设置为不过滤就可以跟oaf共存了，应该是kp把流量抢先接管了所以oaf过滤不了

lxwq, 2021-03-20:
> > > 我用的就是lean的，防火墙fullcone-nat管不管闭这个oaf过滤都不起作用，不知道为什么，内核版本5.4.105
> > 
> > 
> > 广告屏蔽大师 Plus+里的拦截 短视频 APP 和网站是起做用的,不知道为什么,可能是过滤方式和OAF不一样,
> 
> 我知道怎么弄了，kp里访问控制把要屏蔽的设备mac设置为不过滤就可以跟oaf共存了，应该是kp把流量抢先接管了所以oaf过滤不了

广告屏蔽大师 Plus+,ShadowSocksR Plus+之类的软件如果不能同时使用的话是一大遗憾啊,

closed 2021-03-25 by lxwq