destan19/OpenAppFilter #100: 建议发布全平台 all-ipk 文件

JSZMonkey, 2021-03-17:
建议发布全平台 all-ipk 文件，不然每次更新都要去重新编译固件时间太长了。

JSZMonkey, 2021-03-17:
Wed Mar 17 15:35:23 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:35:23 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:35:23 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:03 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:03 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:03 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:12 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:12 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:12 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:20 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:20 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:20 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:23 2021 daemon.err insmod: module is already loaded - oaf
Wed Mar 17 15:36:23 2021 kern.warn kernel: [  459.995155] mac format error: :
Wed Mar 17 15:36:23 2021 daemon.err oafd[22999]: Failed to connect to ubus
Wed Mar 17 15:36:27 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:27 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:27 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:28 2021 daemon.err oafd[23471]: Failed to connect to ubus
Wed Mar 17 15:36:33 2021 daemon.err oafd[24111]: Failed to connect to ubus
Wed Mar 17 15:36:34 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:34 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:34 2021 daemon.err uhttpd[1127]: uci: Entry not found
Wed Mar 17 15:36:38 2021 daemon.err oafd[25199]: Failed to connect to ubus
Wed Mar 17 15:36:43 2021 daemon.err oafd[25210]: Failed to connect to ubus
Wed Mar 17 15:36:48 2021 daemon.err oafd[25211]: Failed to connect to ubus
Wed Mar 17 15:36:48 2021 daemon.info procd: Instance appfilter::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

JSZMonkey, 2021-03-17:
特征库上传之后不更新 还是旧版

yzxj3180, 2021-03-19:
强烈附议

destan19, 2021-03-20:
库上的特征库目前就是最新版本的

closed 2021-03-20 by destan19