destan19/OpenAppFilter #102: 官方master分支下更新到最新版本后仍然不能显示用户也不显示访问记录！！

zxlhhyccc, 2021-03-21:
已经5、6个小时了。。。。
[image]

keeno1982, 2021-03-22:
什么时候能修复呢？

zxlhhyccc, 2021-03-22:
> 什么时候能修复呢？

只有等dalao修复了，我菜鸟。。。。

zxlhhyccc, 2021-03-22:
已经快两天了，还是没显示用户和访问记录：
[image]

lxwq, 2021-03-23:
[image]
为什么我用官方master分支编译后打不开,出这个错误

zxlhhyccc, 2021-03-23:
> [image]
> 为什么我用官方master分支编译后打不开,出这个错误

oaf是稳定版还是最新版？稳定版的luci有个bug，将~=修改我==就可以了。。。你看一下我发的issue

keeno1982, 2021-03-29:
@zxlhhyccc 你那边可以显示用户了吗？

zxlhhyccc, 2021-04-02:
> @zxlhhyccc 你那边可以显示用户了吗？

不能。。。。

secretandanon, 2021-04-10:
看这个 https://github.com/destan19/OpenAppFilter/issues/112

closed 2022-07-19 by destan19