destan19/OpenAppFilter #104: 启动oafd报错

d9394, 2021-03-28:
昨天自已编译的，通过git下下载最新的master，openwrt版本是19.07，路由器是newifi d2(7621)
编译出来kmod-oaf_4.14.224%2B5.0-1_mipsel_24kc.ipk，但downloads.openwrt.org上19.07.7内核只是4.14.221，我想内核版本号一个是221一个是224，应该相差不大吧，用了--force-depends强行安装，但启动时报下面的错：

root@openwrt:/usr/bin# ./oafd
init appfilter
class id = 1, class name = 聊天
class id = 2, class name = 游戏
class id = 3, class name = 视频
class id = 4, class name = 购物
class id = 5, class name = 音乐
class id = 6, class name = 招聘
class id = 7, class name = 下载
class id = 8, class name = 常用网站
Connect netlink 29 failed Protocol not supportedroot@openwrt:/usr/bin#

destan19, 2021-04-02:
强制安装内核模块后，把oaf.ko模块拷贝到系统内核目录，默认安装是以插件包的版本号来安装的，insmod oaf会找不到文件导致模块加载失败，模块路径/lib/modules/4.14.xx/oaf.ko

d9394, 2021-04-05:
root@openwrt:~# opkg install appfilter_3.0-1_mipsel_24kc.ipk
Installing appfilter (3.0-1) to root...
Configuring appfilter.
#class chat
#class game
#class video
#class shopping
#class music
#class employee
#class download
#class website
ok
clean appfilter rule...
enable =
appfilter is disabled
/usr/bin/appfilter.sh: line 37: can't create /proc/sys/oaf/enable: nonexistent directory
appid list=
appid=:
json str={ "op": 4, "data": { "mac_list": [ ":" ] } }

重新编译3.0后，安装appfilter有个错

d9394, 2021-04-05:
试了把/lib/modules/4.14.224/oaf.ko 拷到/lib/modules/4.14.221/
启动好像是正常了：
root@openwrt:~# dmesg | grep oaf
[   14.686568] init oaf sysctl...ok
root@openwrt:~# logread | grep oaf
Mon Apr  5 18:44:29 2021 kern.warn kernel: [   14.686568] init oaf sysctl...ok
但是好像没法显示访问记录，过滤也不起作用
[image]

d9394, 2021-04-09:
logread中出现：
Fri Apr  9 23:38:33 2021 kern.warn kernel: [343957.879560] mac format error: :
Fri Apr  9 23:38:38 2021 daemon.err uhttpd[1315]: uci: Entry not found
Fri Apr  9 23:38:38 2021 daemon.err uhttpd[1315]: uci: Entry not found
Fri Apr  9 23:38:38 2021 daemon.err uhttpd[1315]: uci: Entry not found
是什么问题？

secretandanon, 2021-04-10:
那个是正常的，如果你的无法生效很有可能是LAN口没有桥接，请看说明https://github.com/destan19/OpenAppFilter/issues/112

d9394, 2021-04-23:
[image]

> 
> 
> 那个是正常的，如果你的无法生效很有可能是LAN口没有桥接，请看说明#112

不可能的，只要路由器有LAN的WIFI，自然需要桥接

nnssz, 2022-03-08:
> 

👍

closed 2022-07-19 by destan19