destan19/OpenAppFilter #105: 运行环境检测失败，请先关闭ACC加速模块!

xiaoqi2020, 2021-03-29:
实际上并没有使用ACC加速模块，编译固件的时候也没有选择加速模块
[image]

xylz0928, 2021-03-30:
同样问题

zxlhhyccc, 2021-03-31:
> 实际上并没有使用ACC加速模块，编译固件的时候也没有选择加速模块
> [image]

默认肯定是加速模块已编译。。。

xiaoqi2020, 2021-03-31:
> > 实际上并没有使用ACC加速模块，编译固件的时候也没有选择加速模块
> > [image]
> 
> 默认肯定是加速模块已编译。。。

自己编译的固件，有没有勾选ACC加速模块自己心里还是清楚的，你如果一致认为插件没问题，不可能出现BUG的话，我也没办法！

xiaoqi2020, 2021-03-31:
> > 实际上并没有使用ACC加速模块，编译固件的时候也没有选择加速模块
> > [image]
> 
> 默认肯定是加速模块已编译。。。
所有插件都在这了，
[image]
[image]

destan19, 2021-04-02:
防火墙fullcone-nat需要关闭

xiaoqi2020, 2021-04-03:
> 防火墙fullcone-nat需要关闭

Ok了，谢谢，关闭fullcone-nat后有没有什么别的影响？

secretandanon, 2021-04-10:
目测没有

closed 2022-07-19 by destan19