destan19/OpenAppFilter #110: 配置appid=1002后无法禁用微信聊天，日志显示大量drop 1002了但是实际还是可以聊天，配置appid=3006后，微信公众号也可以正常访问就是加载会稍微慢一点

truelv, 2021-04-07:
2021.04.07-15:31:45  warn  kernel: [86917.210000] [4] 192.168.18.13 visit 1002, time=1617780705 action=Drop, 1866/1869
2021.04.07-15:31:45  info  kernel: [86917.220000] ##drop appid = 1002
2021.04.07-15:31:45  info  kernel: [86917.220000] 
2021.04.07-15:31:45  info  kernel: [86917.220000] 
2021.04.07-15:31:45  warn  kernel: [86917.340000] match by pos, appid=1002
2021.04.07-15:31:45  info  kernel: [86917.340000] match node->ip=192.168.18.13, ip=192.168.18.13
2021.04.07-15:31:45  warn  kernel: [86917.360000] __af_update_client_app_info 735 visit_app_num = 0
2021.04.07-15:31:45  warn  kernel: [86917.370000] update time = 1617780705
2021.04.07-15:31:45  warn  kernel: [86917.370000] [4] 192.168.18.13 visit 1002, time=1617780705 action=Drop, 1867/1870
2021.04.07-15:31:45  info  kernel: [86917.390000] ##drop appid = 1002
2021.04.07-15:31:45  info  kernel: [86917.390000] 
2021.04.07-15:31:45  info  kernel: [86917.390000] 
2021.04.07-15:31:46  warn  kernel: [86917.540000] match by pos, appid=1002
2021.04.07-15:31:46  info  kernel: [86917.550000] match node->ip=192.168.18.13, ip=192.168.18.13
2021.04.07-15:31:46  warn  kernel: [86917.560000] __af_update_client_app_info 735 visit_app_num = 0
2021.04.07-15:31:46  warn  kernel: [86917.570000] update time = 1617780706
2021.04.07-15:31:46  warn  kernel: [86917.580000] [4] 192.168.18.13 visit 1002, time=1617780706 action=Drop, 1868/1871
2021.04.07-15:31:46  info  kernel: [86917.590000] ##drop appid = 1002
2021.04.07-15:31:46  info  kernel: [86917.590000] 
2021.04.07-15:31:46  info  kernel: [86917.590000] 
2021.04.07-15:31:46  warn  kernel: [86917.680000] match by pos, appid=1002
2021.04.07-15:31:46  info  kernel: [86917.680000] match node->ip=192.168.18.13, ip=192.168.18.13
2021.04.07-15:31:46  warn  kernel: [86917.700000] __af_update_client_app_info 735 visit_app_num = 0
2021.04.07-15:31:46  warn  kernel: [86917.710000] update time = 1617780706
2021.04.07-15:31:46  warn  kernel: [86917.710000] [4] 192.168.18.13 visit 1002, time=1617780706 action=Drop, 1869/1872
2021.04.07-15:31:46  info  kernel: [86917.730000] ##drop appid = 1002

truelv, 2021-04-07:
使用的最新版本feature库（v21.4.2）

truelv, 2021-04-07:
没有网络加速和去广告类插件，过滤相关的模块只有oaf这一个

destan19, 2021-04-08:
应该是特征库问题，后面看看

closed 2022-07-19 by destan19