destan19/OpenAppFilter #112: [安装说明] 如何正确的安装OpenAppFilter应用过滤插件

secretandanon, 2021-04-10:
### 首先感谢开发者的无私奉献。
我看开发者并没有详细说明如何安装，而我在安装时遇到了重重问题，折腾一整天才解决，为了不让大家有同样的疑问，现拿出来给大家说道说道，忘大家少走弯路，大家有什么问题也请在下面提出，好一起讨论！

因为这个应用过滤插件设计到很底层，如进程间通信、netfilter（防火墙）、等，所以内核版本如果不对的话很有可能安装不上，强烈建议大家用编译的方式来安装。
下面说一下大致过程：
1. 首先在ubuntu 20.04LTS上克隆lean的源，（我用的是lean的源）
2. 用nano编辑 源根目录下的feeds.default.conf，添加一行
`src-git openappfilter https://github.com/destan19/OpenAppFilter`
3. 然后按照lean的说明进行编译，在编译过程中记得选luci-app-oaf
4. 编译好之后，将固件安装到路由器/软路由中
5. 将冲突的插件关闭，包括Turbo ACC，fullcone-nat（在luci界面的防火墙中关闭）等
6. 看一下网络接口，确保LAN口是桥接的状态（LAN口如果只有一个也要选成桥接）
7. 开启应用过滤，大概1min之后就能收集到数据
8. 如果1min之后没收集到，一定是有问题，不要再等了
9. 注意这个插件不能和不可描述插件共用，因为这个插件底层是通过过滤FORWARD上的包，但那些不可描述插件很多在PREROUTING就已经转发了，所以没办法共用；但如果走tun的话可以与这款插件共用。

zxlhhyccc, 2021-04-14:
> ### 首先感谢开发者的无私奉献。
> 我看开发者并没有详细说明如何安装，而我在安装时遇到了重重问题，折腾一整天才解决，为了不让大家有同样的疑问，现拿出来给大家说道说道，忘大家少走弯路，大家有什么问题也请在下面提出，好一起讨论！
> 
> 因为这个应用过滤插件涉及到很底层，如进程间通信、netfilter（防火墙）、等，所以内核版本如果不对的话很有可能安装不上，强烈建议大家用编译的方式来安装。
> 下面说一下大致过程：
> 
> 1. 首先在ubuntu 20.04LTS上克隆lean的源，（我用的是lean的源）
> 2. 用nano编辑 源根目录下的feeds.default.conf，添加一行
>    `src-git openappfilter https://github.com/destan19/OpenAppFilter`
> 3. 然后按照lean的说明进行编译，在编译过程中记得选luci-app-oaf
> 4. 编译好之后，将固件安装到路由器/软路由中
> 5. 将冲突的插件关闭，包括Turbo ACC，fullcone-nat（在luci界面的防火墙中关闭）等
> 6. 看一下网络接口，确保LAN口是桥接的状态（LAN口如果只有一个也要选成桥接）
> 7. 开启应用过滤，大概1min之后就能收集到数据
> 8. 如果1min之后没收集到，一定是有问题，不要再等了
> 9. 注意这个插件不能和不可描述插件共用，因为这个插件底层是通过过滤FORWARD上的包，但那些不可描述插件很多在PREROUTING就已经转发了，所以没办法共用；但如果走tun的话可以与这款插件共用。

请问你是哪个源码分支编译的？我用的是官方master分支编译了，但很多功能无法实现。。。。

secretandanon, 2021-04-18:
就是最新的

closed 2022-07-19 by destan19

nevertoday, 2022-11-06:
对于已经编译好的插件， 能不能直接用？ ？