destan19/OpenAppFilter #115: 能不能加一个自定义的组，顺带发几个自己添加的特征吗已测试可用

icecubi, 2021-04-15:
香肠派对:[tcp;;6001;;;00:55|01:32]
表情包战争:[tcp;;443;emoji.resource.yuanzililiang.cn;;,tcp;;443;emoji.yuanzililiang.cn;;]
机动都市阿尔法:[tcp;;443;drpf-g93.proxima.nie.netease.com;;]
塔读文学:[tcp;;;tadu.com;;]
番茄小说:[tcp;;443;reading.snssdk.com;;,tcp;;443;reading-lq.snssdk.com;;]
腾讯动漫:[tcp;;443;i.ac.qq.com;;]
菠萝包轻小说:[tcp;;443;sfacg.com;;]

wlshdjj, 2021-04-20:
#class user
9001 这是示例:[tcp;;;www.example.com.jp;;]
9002 香肠派对:[tcp;;6001;;;00:55|01:32]
9003 表情包战争:[tcp;;443;emoji.resource.yuanzililiang.cn;;,tcp;;443;emoji.yuanzililiang.cn;;]
9004 机动都市阿尔法:[tcp;;443;drpf-g93.proxima.nie.netease.com;;]
9005 塔读文学:[tcp;;;tadu.com;;]
9006 番茄小说:[tcp;;443;reading.snssdk.com;;,tcp;;443;reading-lq.snssdk.com;;]
9007 腾讯动漫:[tcp;;443;i.ac.qq.com;;]
9008 菠萝包轻小说:[tcp;;443;sfacg.com;;]

wlshdjj, 2021-04-20:
放在同一个文件的后果是不小心升级时会丢失。我的做法是另外做一个文件，启动时一起读入，不过这要修改启动脚本。这个我早千年说了，没什么人有兴趣。好在自己改也不难。怕就怕在作者不知什么时候占用了9000，又会出问题。你又不能设为10000，不然要崩。

wlshdjj, 2021-04-20:
要注意末行一定要有换行符！放在同一个文件的后果是不小心升级时会丢失。我的做法是另外做一个文件，启动时一起读入，不过这要修改启动脚本。这个我早千年说了，没什么人有兴趣。好在自己改也不难。怕就怕在作者不知什么时候占用了9000，又会出问题。你又不能设为10000，不然要崩。另外作者如果搞得官方特征库末行没有换行符，也会出问题的，细节要处理。

closed 2022-07-19 by destan19