destan19/OpenAppFilter #132: 在lean源码下编译失败，kernel version=5.10.43

MedyMa, 2021-06-17:
2021-06-17T07:35:51.8724387Z time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#0.35#0.28#0.83
2021-06-17T07:35:51.8865938Z make[2]: Entering directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/package/libs/libpcap'
2021-06-17T07:35:52.0126719Z mkdir -p /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/dl
2021-06-17T07:35:52.0157421Z SHELL= flock /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/tmp/.libpcap-1.10.0.tar.gz.flock -c '  	/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/scripts/download.pl "/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/dl" "libpcap-1.10.0.tar.gz" "8d12b42623eeefee872f123bd0dc85d535b00df4d42e865f993c40f7bfc92b1e" "" "http://www.us.tcpdump.org/release/" "http://www.tcpdump.org/release/"    '
2021-06-17T07:35:52.0859915Z echo "libpcap" >> /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/libpcap.default.install
2021-06-17T07:35:52.0880953Z make[2]: Leaving directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/package/libs/libpcap'
2021-06-17T07:35:52.0887949Z time: package/libs/libpcap/compile#0.18#0.03#0.21
2021-06-17T07:35:52.1031825Z make[2]: Entering directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/package/community/MentoHUST-OpenWrt-ipk'
2021-06-17T07:35:52.3895557Z echo "mentohust" >> /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/target-aarch64_generic_musl/pkginfo/MentoHUST-OpenWrt-ipk.default.install
2021-06-17T07:35:52.3914991Z make[2]: Leaving directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/package/community/MentoHUST-OpenWrt-ipk'
2021-06-17T07:35:52.3917789Z time: package/community/MentoHUST-OpenWrt-ipk/compile#0.14#0.03#0.29
2021-06-17T07:35:52.4061341Z make[2]: Entering directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/package/community/OpenAppFilter/oaf'
2021-06-17T07:35:52.5547869Z rm -f /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/.built
2021-06-17T07:35:52.5568364Z touch /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/.built_check
2021-06-17T07:35:52.5599740Z make -C "/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.43" KCFLAGS="-fmacro-prefix-map=/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl" HOSTCFLAGS="-O2 -I/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/host/include -I/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/hostpkg/include -I/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/target-aarch64_generic_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Jun 17 04:39:41 2021" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/netatop.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/r8125.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/trelay.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gpio-button-hotplug.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/mtd-rw.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/r8168.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/nat46.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/siit.symvers /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/symvers/gl-mifi-mcu.symvers" KERNELRELEASE=5.10.43 M="/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0" EXTRA_CFLAGS=""  modules
2021-06-17T07:35:52.5615926Z make[3]: Entering directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.43'
2021-06-17T07:35:53.2117346Z   CC [M]  /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/af_client_fs.o
2021-06-17T07:35:53.6678423Z /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/af_client_fs.c: In function 'init_af_client_procfs':
2021-06-17T07:35:53.6682572Z /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/af_client_fs.c:168:64: error: passing argument 4 of 'proc_create' from incompatible pointer type [-Werror=incompatible-pointer-types]
2021-06-17T07:35:53.6684832Z      pde = proc_create(AF_CLIENT_PROC_STR, 0440, net->proc_net, &af_client_fops);
2021-06-17T07:35:53.6685619Z                                                                 ^~~~~~~~~~~~~~~
2021-06-17T07:35:53.6687080Z In file included from /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/af_client_fs.c:19:
2021-06-17T07:35:53.6688905Z ./include/linux/proc_fs.h:109:24: note: expected 'const struct proc_ops *' but argument is of type 'const struct file_operations *'
2021-06-17T07:35:53.6690279Z  struct proc_dir_entry *proc_create(const char *name, umode_t mode, struct proc_dir_entry *parent, const struct proc_ops *proc_ops);
2021-06-17T07:35:53.6691543Z                         ^~~~~~~~~~~
2021-06-17T07:35:53.7303005Z cc1: some warnings being treated as errors
2021-06-17T07:35:53.7348347Z make[4]: *** [/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/af_client_fs.o] Error 1
2021-06-17T07:35:53.7350179Z scripts/Makefile.build:279: recipe for target '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/af_client_fs.o' failed
2021-06-17T07:35:53.7352774Z make[3]: *** [/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0] Error 2
2021-06-17T07:35:53.7354353Z Makefile:1821: recipe for target '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0' failed
2021-06-17T07:35:53.7356707Z make[2]: *** [/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/.built] Error 2
2021-06-17T07:35:53.7358266Z make[3]: Leaving directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.10.43'
2021-06-17T07:35:53.7361989Z Makefile:41: recipe for target '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/oaf-5.0/.built' failed
2021-06-17T07:35:53.7369327Z make[2]: Leaving directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/package/community/OpenAppFilter/oaf'
2021-06-17T07:35:53.7370471Z time: package/community/OpenAppFilter/oaf/compile#1.01#0.28#1.33
2021-06-17T07:35:53.7371216Z     ERROR: package/community/OpenAppFilter/oaf failed to build.
2021-06-17T07:35:53.7371996Z make[1]: *** [package/community/OpenAppFilter/oaf/compile] Error 1
2021-06-17T07:35:53.7375180Z package/Makefile:114: recipe for target 'package/community/OpenAppFilter/oaf/compile' failed
2021-06-17T07:35:53.7378593Z make[1]: Leaving directory '/home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt'
2021-06-17T07:35:53.7402099Z /home/runner/work/NanoPi-R4S/NanoPi-R4S/openwrt/include/toplevel.mk:228: recipe for target 'package/compile' failed
2021-06-17T07:35:53.7402989Z make: *** [package/compile] Error 2
2021-06-17T07:35:53.7506516Z ##[error]Process completed with exit code 2.

MedyMa, 2021-06-17:
大佬，能否支持内核5.10？

db-one, 2021-06-17:
同样的错误

HoldOnBro, 2021-06-18:
x86 下，同样编译失败

bleach1991, 2021-06-19:
同样是lean的源码
r2s,r4s成功编译通过
但是x86下失败

mmeiwenlong, 2021-06-20:
很早就提过了，大神不理这问题

MedyMa, 2021-06-20:
> 很早就提过了，大神不理这#125

难道说放弃5.10了？

closed 2021-06-28 by destan19