destan19/OpenAppFilter #156: 最新更新报未声明错误 oaf/app_filter.c:261:2: error: 'fs' undeclared (first use in this function)

tonyliangli, 2022-06-16:
oaf/app_filter.c文件中的
fs = get_fs();
编译时，fs报未声明，烦请检查。

haiibo, 2022-06-18:
相同问题，我是arm设备编译报错，x86正常编译

Portan, 2022-06-19:
前面加个声明，mm_segment_t fs=get_fs();

chd1998, 2022-06-21:
ake[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.194'
  CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c: In function 'load_feature_buf_from_file':
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:261:2: error: 'fs' undeclared (first use in this function); did you mean 'fp'?
  fs = get_fs();
  ^~
  fp
/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:261:2: note: each undeclared identifier is reported only once for each function it appears in
scripts/Makefile.build:262: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o' failed
make[5]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o] Error 1
Makefile:1734: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf' failed
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.194'
Makefile:38: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built' failed
make[3]: Leaving directory '/workdir/openwrt/package/OpenAppFilter/oaf'
make[4]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf] Error 2
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built] Error 2
time: package/OpenAppFilter/oaf/compile#1.03#0.28#1.30
    ERROR: package/OpenAppFilter/oaf failed to build.
make[2]: *** [package/OpenAppFilter/oaf/compile] Error 1
package/Makefile:114: recipe for target 'package/OpenAppFilter/oaf/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
Error: Process completed with exit code 2.

closed 2022-06-21 by destan19