destan19/OpenAppFilter #168: 官方最新OpenWrt固件（内核5.10.146）用不了，请处理下，谢谢~

Stev00, 2022-10-22:
使用以下三个均不行
2022.02.06-oaf-x86_64-v5.0-1-k5.10.96.tar.gz
2022.02.22-oaf-x86_64-v5.0-1-k5.10.100.tar.gz
2022.03.03-oaf-x86_64-v5.0-1-k5.15.26.tar.gz

系统log如下：
Sun Oct 23 02:14:27 2022 kern.warn kernel: [   64.014620] oaf: Unknown symbol __x86_indirect_alt_call_rax (err -2)
Sun Oct 23 02:14:27 2022 kern.warn kernel: [   64.016517] oaf: Unknown symbol _printk (err -2)
Sun Oct 23 02:14:27 2022 daemon.err insmod: failed to insert .//kmod/lib/modules/5.15.26/oaf.ko
Sun Oct 23 02:14:31 2022 kern.warn kernel: [   68.084946] appfilter version:5.0
Sun Oct 23 02:14:31 2022 kern.warn kernel: [   68.087207] init oaf sysctl...ok
Sun Oct 23 02:14:32 2022 daemon.err insmod: module is already loaded - oaf
Sun Oct 23 02:14:32 2022 daemon.err rmmod: unloading the module failed
Sun Oct 23 02:14:32 2022 daemon.err insmod: module is already loaded - oaf

Stev00, 2022-10-22:
官方最新OpenWrt固件版本：
openwrt-22.03.2-x86-64-generic-squashfs-combined.img

destan19, 2022-10-31:
官方固件需要单独编译ipk， 目前只支持lede分支，后续更新

closed 2023-06-12 by destan19