destan19/OpenAppFilter #178: Openwrt.org下载的官方编译固件，基本无过滤效果

eeff11, 2023-02-22:
友华wr1200路由器，发哥处理器MT7621,
为确保kenel等与OAF最新release的兼容问题官方下载 [22.03.2固件，](https://downloads.openwrt.org/releases/22.03.2/targets/ramips/mt7621/openwrt-22.03.2-ramips-mt7621-youhua_wr1200js-squashfs-sysupgrade.bin)

用主路由模式，默认没有开启那个NAT加速，无其它任何插件，然后解压，顺次安装以下5个。
kmod-oaf
appfilter
luci-compat(if the luci version is 2.0, openwrt 19.07+)
luci-app-oaf
luci-i18n-oaf-zh-cn(Chinese Language Pack, optional)

用无线连入手机，测试如下：
常用网站：用手机浏览器，163的确不能访问，百度，新浪，搜狐，没有效果，照样可以访问。
常用APP：微博的确不能访问，闲鱼，bilibili 无效，照样可以访问。

有线连入路由，测试如下：
常用网站：用浏览器，163的确不能访问，百度，新浪，搜狐，没用效果，照样可以访问。

可以顺利访问的网站，只是起始打开速度慢了，成功打开后就正常，是否大网站CDN了，造成过滤无效？

需要提供system log吗？

改天ath和IPQ的再试试，感觉不太可能是架构平台的问题。

destan19, 2023-05-11:
安装后重启下服务或者设备试试，用户列表中点击mac地址进入详情看有没有app统计信息

jjm2473, 2023-05-15:
是不是不支持Nftables。我在22.03.4上测试，网关模式，可以看到用户列表，但看不到app信息，过滤也无效。Fullcone-NAT也关闭了。

jjm2473, 2023-05-15:
哦，也不是完全没有app信息，有微信的

destan19, 2023-05-23:
检查下上网环境有没有科学上网，如果上网记录都没有，很有可能是因为加速或者代理引起，数据报文没有经过应用过滤处理

jjm2473, 2023-05-26:
没有科学上网，就是官方openwrt 22.03.4

closed 2023-06-12 by destan19

upsangel, 2023-12-15:
> oaf不支持nftables，固件里回退到iptables以后就没问题了。

這個信息很有用，最新 官方openwrt 基本上都是跑 nftables，如果不能支援的話就很尷尬，只能用回舊版本的 官方openwrt ，即使自編譯的也可能用不到了？

jjm2473, 2023-12-15:
@upsangel 不一定要使用老版本openwrt，可以切换到iptables。 参考这个补丁 https://github.com/istoreos/istoreos/commit/ec8367044caa19441721e90de02f41df88231e33

同时，如果你使用upnp，miniupnp也需要修改回iptables版本

odyssey-qiu, 2024-03-10:
> @upsangel 不一定要使用老版本openwrt，可以切换到iptables。 参考这个补丁 [istoreos/istoreos@ec83670](https://github.com/istoreos/istoreos/commit/ec8367044caa19441721e90de02f41df88231e33)
> 
> 同时，如果你使用upnp，miniupnp也需要修改回iptables版本

看代码也没有用到Iptables工具，为什么要切换回Iptbales？nftables也是兼容旧的iptables命令吧

jjm2473, 2024-03-11:
@odyssey-qiu 你说的是oaf还是upnp？oaf是hook了内核的iptables流程，跟命令无关。upnp的话，你自己看看miniupnp的makefile，是不是有iptables和nftables版本，如果upnp不需要区分防火墙的实现，那还弄两个版本干嘛？ https://github.com/openwrt/packages/blob/openwrt-22.03/net/miniupnpd/Makefile

jjm2473, 2024-03-11:
我今天发现了另一个问题，从代码上看，OAF不支持ipv6流量，如果某些应用通过ipv6访问外网，可能就检测不到了。

jjm2473, 2024-03-14:
还有另一个问题，就是不支持LRO，也就是非线性 SKB buffer。 修复补丁 https://github.com/destan19/OpenAppFilter/pull/193

jjm2473, 2024-03-14:
更正下之前的说法，不是nftables影响过滤效果，因为OAF没处理非线性SKB和ipv6导致的。

ody-26, 2024-03-19:
> @odyssey-qiu 你说的是oaf还是upnp？oaf是hook了内核的netfilter流程，跟命令无关。upnp的话，你自己看看miniupnp的makefile，是不是有iptables和nftables版本，如果upnp不需要区分防火墙的实现，那还弄两个版本干嘛？ https://github.com/openwrt/packages/blob/openwrt-22.03/net/miniupnpd/Makefile

hi，我说的是OAF。我在23.05里测试的，是网关模式，过滤了百度，网易等，但是没有效果，我在hook函数入口加了打印，没有任何打印，感觉根本就没有进hook里。

ody-26, 2024-03-19:
@jjm2473 hello,我说的是OAF。我在23.05里测试的，是网关模式，过滤了百度，网易等，但是没有效果，我在hook函数入口加了打印，没有任何打印，感觉根本就没有进hook里。