destan19/OpenAppFilter #179: Openwrt Nightly Build

tango38317, 2023-05-15:
Hello @destan19  ,
i tried to add OAF to my build system (Openwrt Snapshot) and tried to build a nightly build with OAF, but build is not successful.
i receive a view error attached to log.
maybe you have some hints how to solve this issue
many thanks
Chris
[oaf.txt](https://github.com/destan19/OpenAppFilter/files/11477767/oaf.txt)

wzming, 2023-05-18:
[image]

同样的问题，openwrt master分支，oaf master分支，WSL2-UBUNTU

destan19, 2023-05-25:
Hello tango
It is because some kernel compilations have enabled strict compilation mode, causing warnings to be treated as errors, and relevant warnings have been ignored or fixed.
thanks for feedback.

closed 2023-05-25 by destan19