destan19/OpenAppFilter #205: app_filter.c的1267 行在OPENWRT编译时报第9个参数错误，认为应该是int传的是long long unsigned int

zcluo, 2025-02-05:
/home/zcluo/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:40: error: format '%d' expects argument of type 'int', but argument 9 has type 'long long unsigned int' [-Werror=format=]
 1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1268 |                                         &flow.src, flow.sport, &flow.dst, flow.dport, skb->len, flow.app_id, total_packets);
      |                                                                                                              ~~~~~~~~~~~~~
      |                                                                                                              |
      |                                                                                                              long long unsigned int
./include/linux/printk.h:427:25: note: in definition of macro 'printk_index_wrap'
  427 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
      |                         ^~~~
/home/zcluo/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:33: note: in expansion of macro 'printk'
 1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
      |                                 ^~~~~~
/home/zcluo/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:91: note: format string is defined here
 1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
      |                                                                                          ~^
      |                                                                                           |
      |                                                                                           int
      |                                                                                          %lld
cc1: all warnings being treated as errors

zcluo, 2025-02-05:
x86 openwrt master 分支

zcluo, 2025-02-05:
printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n    改成 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %lld \n   编译可以通过。

Andy-Home, 2025-02-05:
我编译时也遇到了这个问题
`2025-02-05T08:21:53.6821779Z make[3]: Entering directory '/workdir/openwrt/package/OpenAppFilter/oaf'
2025-02-05T08:21:53.8194612Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built
2025-02-05T08:21:53.8223395Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built_check
2025-02-05T08:21:53.8249240Z make -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Feb  1 05:19:18 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="" KERNELRELEASE=5.15.173 M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf" EXTRA_CFLAGS="-Wno-declaration-after-statement -Wno-strict-prototypes -Wno-unused-variable -Wno-implicit-fallthrough -Wno-missing-braces -Wno-parentheses"  modules
2025-02-05T08:21:53.8273657Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173'
2025-02-05T08:21:54.1886046Z   CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o
2025-02-05T08:21:54.7460925Z In file included from ./include/linux/kernel.h:19,
2025-02-05T08:21:54.7461580Z                  from ./include/linux/list.h:9,
2025-02-05T08:21:54.7462053Z                  from ./include/linux/module.h:12,
2025-02-05T08:21:54.7462767Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:6:
2025-02-05T08:21:54.7464307Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c: In function 'app_filter_hook_gateway_handle':
2025-02-05T08:21:54.7466061Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:40: error: format '%d' expects argument of type 'int', but argument 9 has type 'long long unsigned int' [-Werror=format=]
2025-02-05T08:21:54.7467648Z  1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
2025-02-05T08:21:54.7468475Z       |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2025-02-05T08:21:54.7469258Z  1268 |                                         &flow.src, flow.sport, &flow.dst, flow.dport, skb->len, flow.app_id, total_packets);
2025-02-05T08:21:54.7470024Z       |                                                                                                              ~~~~~~~~~~~~~
2025-02-05T08:21:54.7470790Z       |                                                                                                              |
2025-02-05T08:21:54.7471366Z       |                                                                                                              long long unsigned int
2025-02-05T08:21:54.7472213Z ./include/linux/printk.h:422:25: note: in definition of macro 'printk_index_wrap'
2025-02-05T08:21:54.7472945Z   422 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
2025-02-05T08:21:54.7473460Z       |                         ^~~~
2025-02-05T08:21:54.7474337Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:33: note: in expansion of macro 'printk'
2025-02-05T08:21:54.7475750Z  1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
2025-02-05T08:21:54.7476464Z       |                                 ^~~~~~
2025-02-05T08:21:54.7477380Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:91: note: format string is defined here
2025-02-05T08:21:54.7478998Z  1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
2025-02-05T08:21:54.7480087Z       |                                                                                          ~^
2025-02-05T08:21:54.7480625Z       |                                                                                           |
2025-02-05T08:21:54.7481151Z       |                                                                                           int
2025-02-05T08:21:54.7481677Z       |                                                                                          %lld
2025-02-05T08:21:55.1050988Z cc1: all warnings being treated as errors
2025-02-05T08:21:55.1152421Z make[5]: *** [scripts/Makefile.build:289: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o] Error 1
2025-02-05T08:21:55.1154053Z make[4]: *** [Makefile:1907: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf] Error 2
2025-02-05T08:21:55.1155371Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173'
2025-02-05T08:21:55.1156702Z make[3]: *** [Makefile:45: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built] Error 2
[... cut ...]
2025-02-05T08:21:55.1174389Z make[2]: *** [package/Makefile:129: package/OpenAppFilter/oaf/compile] Error 1
2025-02-05T08:21:55.1179263Z make[2]: Leaving directory '/workdir/openwrt'
2025-02-05T08:21:55.1210138Z make[1]: *** [package/Makefile:123: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2025-02-05T08:21:55.1214545Z make[1]: Leaving directory '/workdir/openwrt'
2025-02-05T08:21:55.1246481Z make: *** [/workdir/openwrt/include/toplevel.mk:233: world] Error 2
2025-02-05T08:21:55.1268546Z ##[error]Process completed with exit code 2.`

Andy-Home, 2025-02-05:
编译时遇到该问题
2025-02-05T08:21:53.6821779Z make[3]: Entering directory '/workdir/openwrt/package/OpenAppFilter/oaf'
2025-02-05T08:21:53.8194612Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built
2025-02-05T08:21:53.8223395Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built_check
2025-02-05T08:21:53.8249240Z make -C "/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Feb  1 05:19:18 2025" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="" KERNELRELEASE=5.15.173 M="/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf" EXTRA_CFLAGS="-Wno-declaration-after-statement -Wno-strict-prototypes -Wno-unused-variable -Wno-implicit-fallthrough -Wno-missing-braces -Wno-parentheses"  modules
2025-02-05T08:21:53.8273657Z make[4]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173'
2025-02-05T08:21:54.1886046Z   CC [M]  /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o
2025-02-05T08:21:54.7460925Z In file included from ./include/linux/kernel.h:19,
2025-02-05T08:21:54.7461580Z                  from ./include/linux/list.h:9,
2025-02-05T08:21:54.7462053Z                  from ./include/linux/module.h:12,
2025-02-05T08:21:54.7462767Z                  from /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:6:
2025-02-05T08:21:54.7464307Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c: In function 'app_filter_hook_gateway_handle':
2025-02-05T08:21:54.7466061Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:40: error: format '%d' expects argument of type 'int', but argument 9 has type 'long long unsigned int' [-Werror=format=]
2025-02-05T08:21:54.7467648Z  1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
2025-02-05T08:21:54.7468475Z       |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2025-02-05T08:21:54.7469258Z  1268 |                                         &flow.src, flow.sport, &flow.dst, flow.dport, skb->len, flow.app_id, total_packets);
2025-02-05T08:21:54.7470024Z       |                                                                                                              ~~~~~~~~~~~~~
2025-02-05T08:21:54.7470790Z       |                                                                                                              |
2025-02-05T08:21:54.7471366Z       |                                                                                                              long long unsigned int
2025-02-05T08:21:54.7472213Z ./include/linux/printk.h:422:25: note: in definition of macro 'printk_index_wrap'
2025-02-05T08:21:54.7472945Z   422 |                 _p_func(_fmt, ##__VA_ARGS__);                           \
2025-02-05T08:21:54.7473460Z       |                         ^~~~
2025-02-05T08:21:54.7474337Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:33: note: in expansion of macro 'printk'
2025-02-05T08:21:54.7475750Z  1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
2025-02-05T08:21:54.7476464Z       |                                 ^~~~~~
2025-02-05T08:21:54.7477380Z /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.c:1267:91: note: format string is defined here
2025-02-05T08:21:54.7478998Z  1267 |                                 printk(" %s %pI4(%d)--> %pI4(%d) len = %d, %d ,pkt num = %d \n ", IPPROTO_TCP == flow.l4_protocol ? "tcp" : "udp",
2025-02-05T08:21:54.7480087Z       |                                                                                          ~^
2025-02-05T08:21:54.7480625Z       |                                                                                           |
2025-02-05T08:21:54.7481151Z       |                                                                                           int
2025-02-05T08:21:54.7481677Z       |                                                                                          %lld
2025-02-05T08:21:55.1050988Z cc1: all warnings being treated as errors
2025-02-05T08:21:55.1152421Z make[5]: *** [scripts/Makefile.build:289: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/app_filter.o] Error 1
2025-02-05T08:21:55.1154053Z make[4]: *** [Makefile:1907: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf] Error 2
2025-02-05T08:21:55.1155371Z make[4]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.15.173'
2025-02-05T08:21:55.1156702Z make[3]: *** [Makefile:45: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.built] Error 2
[... cut ...]
2025-02-05T08:21:55.1174389Z make[2]: *** [package/Makefile:129: package/OpenAppFilter/oaf/compile] Error 1
2025-02-05T08:21:55.1179263Z make[2]: Leaving directory '/workdir/openwrt'
2025-02-05T08:21:55.1210138Z make[1]: *** [package/Makefile:123: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2025-02-05T08:21:55.1214545Z make[1]: Leaving directory '/workdir/openwrt'
2025-02-05T08:21:55.1246481Z make: *** [/workdir/openwrt/include/toplevel.mk:233: world] Error 2
2025-02-05T08:21:55.1268546Z ##[error]Process completed with exit code 2.

Andy-Home, 2025-02-05:
编译版本为23.05

destan19, 2025-02-07:
已经修复

closed 2025-02-07 by destan19