destan19/OpenAppFilter #206: g_af_config重复定义编译失败

zcluo, 2025-02-07:
x86_64-openwrt-linux-musl-gcc -o oafd appfilter_user.o appfilter_netlink.o appfilter_ubus.o appfilter_config.o utils.o main.o -L/home/zcluo/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/zcluo/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lm -lpthread -lubox -luci -lubus -ljson-c -lblobmsg_json
/home/zcluo/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: appfilter_netlink.o:(.bss+0x0): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
/home/zcluo/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: appfilter_ubus.o:(.bss+0x20): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
/home/zcluo/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: appfilter_config.o:(.bss+0x11220): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
/home/zcluo/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: main.o:(.bss+0x20): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here

MilesPoupart, 2025-02-07:
我也遇到了
```
make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/package/community/OpenAppFilter/open-app-filter'
rm -f /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter/.built
touch /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter/.built_check
make -C /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter/ CC="x86_64-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter=appfilter -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LIBS="-L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lm -lpthread -lubox -luci -lubus -ljson-c -lblobmsg_json" all
make[3]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter'
x86_64-openwrt-linux-musl-gcc -o oafd appfilter_user.o appfilter_netlink.o appfilter_ubus.o appfilter_config.o utils.o main.o -L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/usr/lib -L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib -fuse-ld=bfd -znow -zrelro -lm -lpthread -lubox -luci -lubus -ljson-c -lblobmsg_json
/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: appfilter_netlink.o:(.bss+0x0): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: appfilter_ubus.o:(.bss+0x20): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: appfilter_config.o:(.bss+0x11220): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/toolchain-x86_64_gcc-13.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld.bfd: main.o:(.bss+0x20): multiple definition of `g_af_config'; appfilter_user.o:(.bss+0x220): first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:4: all] Error 1
make[3]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter'
make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/package/community/OpenAppFilter/open-app-filter'
make[2]: *** [Makefile:58: /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/appfilter/.built] Error 2
time: package/community/OpenAppFilter/open-app-filter/compile#0.09#0.00#0.08
    ERROR: package/community/OpenAppFilter/open-app-filter failed to build.
make[1]: *** [package/Makefile:116: package/community/OpenAppFilter/open-app-filter/compile] Error 1
make[1]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt'
make: *** [/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/include/toplevel.mk:231: package/compile] Error 2
```

destan19, 2025-02-07:
已经修复

closed 2025-02-07 by destan19