destan19/OpenAppFilter #207: 新版的是不是对IPV6的不起作用？

gyangcn, 2025-02-10:
新版的是不是对IPV6的不起作用？

cnwikee, 2025-02-10:
> 新版的是不是对IPV6的不起作用？

你能正常编译吗？我编译出来，没有kmod-oaf.ipk文件，导致安装 luci-app-oaf_6.1.1_all.ipk 时候报错
```base Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-oaf for luci-app-oaf
 * opkg_download_pkg: Package luci-app-oaf is not available from any configured src.
 * opkg_install_pkg: Failed to download luci-app-oaf. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci-app-oaf.```

qqsir-dev, 2025-02-10:
我能编译新版的，就是不生效，主路由，网关模式，我是默认开启了ipv6，不知道是不是这个原因

cnwikee, 2025-02-10:
> 我能编译新版的，就是不生效，主路由，网关模式，我是默认开启了ipv6，不知道是不是这个原因

哈哈，我找到了，编译后的路径在：openwrt-sdk/bin/targets/x86/64/packages/kmod-oaf_5.10.221-1_x86_64.ipk，不在openwrt-sdk/bin/packages/x86_64/base里面。

CXH404, 2025-02-12:
软件在23.05.4源码下编译成功，和前面的朋友一样没有生成kmod-oaf。其余三个安顺序安装后，在软件包已安装中看不到luci-app-oaf，但能看到软件界面，只是所有修改无法保存。

gyangcn, 2025-02-12:
不是ipv6的问题，打开了openclash如果不开启绕经中国大陆模式就不起作用，打开这个模式就起作用了

cnwikee, 2025-02-13:
> 软件在23.05.4源码下编译成功，和前面的朋友一样没有生成kmod-oaf。其余三个安顺序安装后，在软件包已安装中看不到luci-app-oaf，但能看到软件界面，只是所有修改无法保存。

编译后的路径在：openwrt-sdk/bin/targets/x86/64/packages/kmod-oaf_5.10.221-1_x86_64.ipk，不在openwrt-sdk/bin/packages/x86_64/base里面。或者你搜索一下kmod-oaf，是编译出来了的，只是不在默认路径，我安装好了，没问题。3.0的特征库可以按说明自定义修改。