destan19/OpenAppFilter #209: 新版编译后，应用过虑功能异常

bintrees, 2025-02-13:
应用列表勾选无反应，变更【工作模式】与打开关闭【应用过虑】保存提示成功，但刷新后发现未保存；

[image]

[image]

qqsir-dev, 2025-02-13:
是按照官方编译的要求做的吗？如果是替换APP应该是要有东西没有被替换
建议删除源里面带有luci-app-oaf和open-app-filter的包，然后clone官方整体，包含oaf，这样才完整

bintrees, 2025-02-14:
> 是按照官方编译的要求做的吗？如果是替换APP应该是要有东西没有被替换 建议删除源里面带有luci-app-oaf和open-app-filter的包，然后clone官方整体，包含oaf，这样才完整

是

> 是按照官方编译的要求做的吗？如果是替换APP应该是要有东西没有被替换 建议删除源里面带有luci-app-oaf和open-app-filter的包，然后clone官方整体，包含oaf，这样才完整

是按这个步骤编译的：
先编译完整固件
git clone https://github.com/coolsnowwolf/lede 
cd lede
git clone https://github.com/destan19/OpenAppFilter.git package/OpenAppFilter （旧的版本）

./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make -j8 download
make -j1 V=s
这个版本oaf是正常的，openwrt版本R24.5.1，内核5.15.158

现在编译升级oaf 

2、移除原oaf，重新clone新版本
cd lede
git clone https://github.com/destan19/OpenAppFilter.git package/OpenAppFilter

echo "CONFIG_PACKAGE_luci-app-oaf=y" >>.config  
     make defconfig
make -j1 V=s

3、升级固件

不知是不是选了保存配置原因

cnwikee, 2025-02-14:
我也是从5.0.2升级过来的，刚开始是覆盖安装的，完成后跟你遇到的情况一样。后来我完整的卸载了kmod-oaf、appfilter、luci-app-oaf、luci-i18n-oaf-zh-cn；然后搜索删除了所有appfilter及包含oaf的文件，再从新安装最新编译的 IPK，正常了。

qqsir-dev, 2025-02-15:
> > 是按照官方编译的要求做的吗？如果是替换APP应该是要有东西没有被替换 建议删除源里面带有luci-app-oaf和open-app-filter的包，然后clone官方整体，包含oaf，这样才完整
> 
> 是
> 
> > 是按照官方编译的要求做的吗？如果是替换APP应该是要有东西没有被替换 建议删除源里面带有luci-app-oaf和open-app-filter的包，然后clone官方整体，包含oaf，这样才完整
> 
> 是按这个步骤编译的： 先编译完整固件 git clone https://github.com/coolsnowwolf/lede cd lede git clone https://github.com/destan19/OpenAppFilter.git package/OpenAppFilter （旧的版本）
> 
> ./scripts/feeds update -a ./scripts/feeds install -a make menuconfig make -j8 download make -j1 V=s 这个版本oaf是正常的，openwrt版本R24.5.1，内核5.15.158
> 
> 现在编译升级oaf
> 
> 2、移除原oaf，重新clone新版本 cd lede rm -rf package/OpenAppFilter git clone https://github.com/destan19/OpenAppFilter.git package/OpenAppFilter
> 
> echo "CONFIG_PACKAGE_luci-app-oaf=y" >>.config make defconfig make -j1 V=s
> 
> 3、升级固件
> 
> 不知是不是选了保存配置原因

删除源里面的luci-app-appfilter， oaf， OpenAppFilter，好像你只删除了后两样

hondahybird, 2025-02-16:
同样的问题，重置系统重新刷的包。

closed 2025-02-17 by bintrees