destan19/OpenAppFilter #214: kmod-oaf无法安装

stevenliuit, 2025-02-24:
系统为istoreos 用的r68s的硬件
本来系统里有 默认给卸载了 然后安装下的包提示下面错误

 opkg install kmod-oaf_5.10.221-1_aarch64_generic.ipk 
Unknown package 'kmod-oaf'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.221-1-1627d9921c1ff5b7468a1701dee0ac85) for kmod-oaf
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-oaf found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-oaf.
内核版本
uname -a
Linux iStoreOS 5.10.221 #0 SMP Tue Dec 24 12:44:43 2024 aarch64 GNU/Linux

系统版本iStoreOS 22.03.7 2024122712 / LuCI istoreos-22.03 branch git-24.339.46321-5f9267c
下载的包：[oaf_v6.1.1-rockchip_armv8-kernel_5.10.221.tar.gz](http://111.230.32.167:8086/files/ipk/openwrt/rockchip_armv8/openwrt-22.03/oaf_v6.1.1-rockchip_armv8-kernel_5.10.221.tar.gz)
求大佬指导

tian4546, 2025-02-25:
24.10.0版本软件中心有oaf驱动直接安装

stevenliuit, 2025-02-25:
> 24.10.0版本软件中心有oaf驱动直接安装

我去搜搜

stevenliuit, 2025-02-25:
> 注意看说明文档，驱动不能直接通过ipk安装，要手动insmod

忘了看前面文档结果。。。。。。。

stevenliuit, 2025-02-25:
> 24.10.0版本软件中心有oaf驱动直接安装

[image]
我的还是这个版本 istoreos没更新

stevenliuit, 2025-02-25:
> 24.10.0版本软件中心有oaf驱动直接安装

[image]

[image]
这两个都搜不到

stevenliuit, 2025-02-25:
> 注意看说明文档，驱动不能直接通过ipk安装，要手动insmod

[image]
我看的文档的文章 确实 执行 还是无法安装 下面是我参考的文章地址
http://doc.openappfilter.com/oaf/ipk-install.html#_7-%E5%AE%89%E8%A3%85%E6%8F%92%E4%BB%B6

stevenliuit, 2025-02-25:
> 注意看说明文档，驱动不能直接通过ipk安装，要手动insmod

[image]

安装完了 也是提示未运行(oaf驱动未加载,请下载该固件对应内核版本的oaf.ko驱动文件进行安装，并开启自动加载，当前内核版本为5.10.221)

root@iStoreOS:/tmp/oaf_v6.1.1-rockchip_armv8-kernel_5.10.221/other_ipks# opkg install kmod-oaf_5.10.221-1_aarch64_generic.ipk 
Unknown package 'kmod-oaf'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.221-1-1627d9921c1ff5b7468a1701dee0ac85) for kmod-oaf
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-oaf found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-oaf.

安装这个kmod-oaf还是报错这个
还有一个就是 我看了拷贝路径下面还有个内核路径 但是我里面拷贝进去了一个 不是mv的 也不行 还是提示这个 怀疑是未能安装kmod-oaf导致的

[image]

stevenliuit, 2025-02-25:
> [image]
>
> 安装完了 也是提示未运行(oaf驱动未加载,请下载该固件对应内核版本的oaf.ko驱动文件进行安装，并开启自动加载，当前内核版本为5.10.221)
>
> root@iStoreOS:/tmp/oaf_v6.1.1-rockchip_armv8-kernel_5.10.221/other_ipks# opkg install [...]

[image]
提示也挂载了

stevenliuit, 2025-02-25:
[image]

[image]
这个是安装的包 没有kmod-oaf 卸载了就装不回去了

tian4546, 2025-02-25:
软件包里面有 oaf 驱动，

[image]  搜索出来直接点击安装就可以了，但是我不知道你的 软件包 中心有没有，作者压缩包里面的那个 oaf 需要特定对应编译固件版本和内核号才能安装

stevenliuit, 2025-02-25:
> 软件包里面有 oaf 驱动，
> 
> <img alt="Image" width="1451" src="https://private-user-images.githubusercontent.com/14542275/416466777-f4d55fdf-3442-4272-a83f-f9fed7fd9b3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDA0NTI0NzAsIm5iZiI6MTc0MDQ1MjE3MCwicGF0aCI6Ii8xNDU0MjI3NS80MTY0NjY3NzctZjRkNTVmZGYtMzQ0Mi00MjcyLWE4M2YtZjlmZWQ3ZmQ5YjNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjI1VDAyNTYxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJlMzc3NmQ4NmExOWI3ZGZkNjY0MDk0OGEzNjNhOTA1OGUyMGJkYzZmZmRkYjQ0MmQ2YTgwZTM3ZjQxY2Q3NjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.uI-L_ehVvagSpU0-6EmN1NOW_jhpjn9u3fqETQa8E58"> 搜索出来直接点击安装就可以了，但是我不知道你的 软件包 中心有没有，作者压缩包里面的那个 oaf 需要特定对应编译固件版本和内核号才能安装

我的包里搜不到

[image]

已安装里的是我拷贝进去的

[image]
我的源地址是
src/gz openwrt_base https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/base
src/gz openwrt_luci https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/luci
src/gz openwrt_packages https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/packages
src/gz openwrt_routing https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/routing
src/gz openwrt_telephony https://mirrors.cernet.edu.cn/openwrt/releases/22.03.7/packages/aarch64_cortex-a53/telephony
太难了

tian4546, 2025-02-25:
我的是 手动上传的安装包 同样也安装 不了 报错fail instal NULL 什么的不太记得了，然后去软件包管理中心一搜索居然有，就可以安装

stevenliuit, 2025-02-25:
> 我的是 手动上传的安装包 同样也安装 不了 报错fail instal NULL 什么的不太记得了，然后去软件包管理中心一搜索居然有，就可以安装

你的那个应该是x86的机器 然后是24版本的openwrt 我的是22.03 估计是这个原因

TeddyXiong79, 2025-03-04:
应该是ARMv8的适配有问题。我重做了istoreos-22.03.7-2024122712，发现kmod-oaf是已经安装好的，我就按照文档，仅仅协助了：
opkg remove luci-i18n-oaf-zh-cn
opkg remove luci-app-oaf
opkg remove appfilter
（kmod-oaf保留了系统自带的）
然后后续的安装非常顺利，页面显示也正常的。然后我过了一段时间看看，发现驱动无法加载，我就把kmod-oaf全部卸载了，准备重新装一遍，这下哦豁了。。。。怎么都装不上了。

报错如下：

opkg install kmod-oaf_5.10.221-1_aarch64_generic.ipk 

Unknown package 'kmod-oaf'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.221-1-0fa6cafe250be2c804f4735570c88345) for kmod-oaf
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-oaf found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-oaf.

stevenliuit, 2025-03-07:
> 应该是ARMv8的适配有问题。我重做了istoreos-22.03.7-2024122712，发现kmod-oaf是已经安装好的，我就按照文档，仅仅协助了： opkg remove luci-i18n-oaf-zh-cn opkg remove luci-app-oaf opkg remove appfilter （kmod-oaf保留了系统自带的） [...]

我已经问了istoreos的作者 作者说 这个模块不能卸载 如果卸载了就装不了了 除非重置才能把这个模块装回来 

[image]
顺便大佬也说了 更新系统也出不来 只能重置才行

[image]

TeddyXiong79, 2025-03-08:
这。。。。。
你看看官方的手册里面这么说的：kmod-oaf oaf驱动插件（oaf驱动如果已经存在，可以先不要卸载重装，如果确定不能兼容，再卸载重装）

http://doc.openappfilter.com/oaf/ipk-install.html

结果现在卸载了就装不回去了！！！！

stevenliuit, 2025-03-10:
> 这。。。。。 你看看官方的手册里面这么说的：kmod-oaf oaf驱动插件（oaf驱动如果已经存在，可以先不要卸载重装，如果确定不能兼容，再卸载重装）
> 
> http://doc.openappfilter.com/oaf/ipk-install.html
> 
> 结果现在卸载了就装不回去了！！！！

放弃吧 我重置了之后就好了 

[image]
但是有个问题是 设置启用了 但是还是会失效 限制的app还是能打开使用

closed 2025-03-12 by stevenliuit

mia0x75, 2025-03-18:
GPL 协议，还没公开？
> [image]

slty7, 2025-07-04:
> 这。。。。。 你看看官方的手册里面这么说的：kmod-oaf oaf驱动插件（oaf驱动如果已经存在，可以先不要卸载重装，如果确定不能兼容，再卸载重装）
> 
> http://doc.openappfilter.com/oaf/ipk-install.html
> 
> 结果现在卸载了就装不回去了！！！！

老哥，需要重装吗?还是直接恢复出厂设置就行了