destan19/OpenAppFilter #217: 升级到24.10内核6.6.73仍提示Invalid module format不能安装oaf驱动

Phantom-Blu-Ray, 2025-02-27:
以下为部分日志信息，路由器为小米R3G，从23.05通过sysupgrade升级至24.10
root@OpenWrt:/tmp/oaf_v6.1.1-ramips_mt7621-kernel_6.6.73# uname -r
6.6.73
root@OpenWrt:/tmp/oaf_v6.1.1-ramips_mt7621-kernel_6.6.73# insmod oaf-6.6.73.ko 
insmod: ERROR: could not insert module oaf-6.6.73.ko: Invalid module format
root@OpenWrt:/tmp/oaf_v6.1.1-ramips_mt7621-kernel_6.6.73# dmesg | tail
[  453.784832] module oaf: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time
[  574.003680] module oaf: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time
[  726.890515] module oaf: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time
root@OpenWrt:/tmp/oaf_v6.1.1-ramips_mt7621-kernel_6.6.73# insmod oaf-6.6.73-2.ko 
insmod: ERROR: could not insert module oaf-6.6.73-2.ko: Invalid module format
root@OpenWrt:/tmp/oaf_v6.1.1-ramips_mt7621-kernel_6.6.73#

Phantom-Blu-Ray, 2025-02-27:
这里尝试安装kmod-oaf也显示内核版本似乎有问题？
[image]

Phantom-Blu-Ray, 2025-02-27:
目前通过作者插件官网提供的24.10.0-rc7固件可以正常安装，但是该版本固件不能直接安装24.10.0稳定版的kmod系列组件，希望可以适配稳定版本的

destan19, 2025-03-11:
已经重新发布，官方的24.10.0可以安装

zzduci, 2025-04-03:
> 已经重新发布，官方的24.10.0可以安装

root@ImmortalWrt:/tmp/oaf_v6.1.3-x86_64-kernel_6.6.58# opkg install libubus libuci
Package libubus20250102 (2025.01.02~afa57cce-r1) installed in root is up to date.
Package libuci20250120 (2025.01.20~16ff0bad-r1) installed in root is up to date.
root@ImmortalWrt:/tmp/oaf_v6.1.3-x86_64-kernel_6.6.58# opkg install appfilter_6.1.3-r1_x86_64.ipk
Unknown package 'appfilter'.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubus20241020 for appfilter
 * pkg_hash_check_unresolved: cannot find dependency libuci20130104 for appfilter
 * pkg_hash_fetch_best_installation_candidate: Packages for appfilter found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package appfilter.
root@ImmortalWrt:/tmp/oaf_v6.1.3-x86_64-kernel_6.6.58#   提示缺依赖，但依赖是安装了的，导致IPK安装失败

zzduci, 2025-04-03:
> 已经重新发布，官方的24.10.0可以安装

[image]

zzduci, 2025-04-03:
[image]我手动下载安装，提示这个，我的是immortalwrt 24.10.0

zzduci, 2025-04-03:
[image]

zzduci, 2025-04-03:
我的immortalwrt我把所有的.ko驱动都试了，都不能用，期待早日适配

zzduci, 2025-04-03:
[image]无意之中，在软件包里搜索oaf，结果发现有一个可以安装的，安装完成终于可以正常运行了