destan19/OpenAppFilter #220: NanoPi R5C arm版本安装插件 网页提示未加载

666Robot, 2025-03-02:
iStoreOS

型号 | FriendlyElec NanoPi R5C
架构 | ARMv8 Processor rev 0
目标平台 | rockchip/rk35xx
固件版本 | iStoreOS 22.03.7 2024122712 / LuCI istoreos-22.03 branch git-24.339.46321-5f9267c
内核版本 | 5.10.221
本地时间 | 2025-03-03 05:09:06
运行时间 | 0h 31m 39s
平均负载 | 0.35, 0.38, 0.31

打的这个包oaf_v6.1.1-rockchip_armv8-kernel_5.10.221.tar   我看到你的芯片支持列表确实没有我这个型号 但是近似的型号应该不至于阿 可以帮我调试一下吗？

TeddyXiong79, 2025-03-05:
我的也是一样，驱动死活装不上

TeddyXiong79, 2025-03-05:
我的R5S用iStoreOS，自带的oaf驱动无法加载，我就卸载了准备用这个包oaf_v6.1.1-rockchip_armv8-kernel_5.10.221.tar 里面Other ipk包安装，发现死活安装不上。

opkg install kmod-oaf_5.10.221-1_aarch64_generic.ipk Unknown package 'kmod-oaf'. Collected errors:
pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.221-1-1627d9921c1ff5b7468a1701dee0ac85) for kmod-oaf
pkg_hash_fetch_best_installation_candidate: Packages for kmod-oaf found, but incompatible with the architectures configured
opkg_install_cmd: Cannot install package kmod-oaf.

内核版本
uname -a
Linux iStoreOS 5.10.221 #0 SMP Tue Dec 24 12:44:43 2024 aarch64 GNU/Linux

closed 2026-03-04 by destan19