destan19/OpenAppFilter #222: 在kernel6.6到6.12编译总是失败

wuxeon, 2025-03-05:
如题，由于ipq6000没有公版immortalwrt和openwrt，使用源码（libwrt）编译，
总是报错，请帮忙看看。
  /home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.o
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:59:5: error: no previous prototype for '__add_app_feature' [-Werror=missing-prototypes]
   59 | int __add_app_feature(char *feature, int appid, char *name, int proto, int src_port,
      |     ^~~~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:120:5: error: no previous prototype for 'validate_range_value' [-Werror=missing-prototypes]
  120 | int validate_range_value(char *range_str)
      |     ^~~~~~~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:141:5: error: no previous prototype for 'parse_range_value' [-Werror=missing-prototypes]
  141 | int parse_range_value(char *range_str, range_value_t *range)
      |     ^~~~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:178:5: error: no previous prototype for 'parse_port_info' [-Werror=missing-prototypes]
  178 | int parse_port_info(char *port_str, port_info_t *info)
      |     ^~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:210:5: error: no previous prototype for 'af_match_port' [-Werror=missing-prototypes]
  210 | int af_match_port(port_info_t *info, int port)
      |     ^~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:247:5: error: no previous prototype for 'add_app_feature' [-Werror=missing-prototypes]
  247 | int add_app_feature(int appid, char *name, char *feature)
      |     ^~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:340:6: error: no previous prototype for 'af_init_feature' [-Werror=missing-prototypes]
  340 | void af_init_feature(char *feature_str)
      |      ^~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:393:6: error: no previous prototype for 'load_feature_buf_from_file' [-Werror=missing-prototypes]
  393 | void load_feature_buf_from_file(char **config_buf)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.c:440:5: error: no previous prototype for 'load_feature_config' [-Werror=missing-prototypes]
  440 | int load_feature_config(void)

lean的6.1编译正常，由于6.1内核的内存占用太大，故使用libwrt。

wuxeon, 2025-03-05:
cc1: all warnings being treated as errors
make[5]: *** [scripts/Makefile.build:229: /home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/app_filter.o] Error 1
make[4]: *** [/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.12.16/Makefile:1932: /home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf] Error 2
make[3]: *** [Makefile:224: __sub-make] Error 2
make[3]: Leaving directory '/home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-6.12.16'
make[2]: *** [Makefile:45: /home/ubt2204/openwrt-6.x/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/oaf/.built] Error 2
make[2]: Leaving directory '/home/ubt2204/openwrt-6.x/package/OpenAppFilter/oaf'
time: package/OpenAppFilter/oaf/compile#1.81#0.42#4.65
    ERROR: package/OpenAppFilter/oaf failed to build.
make[1]: *** [package/Makefile:185: package/OpenAppFilter/oaf/compile] Error 1

destan19, 2025-03-11:
最新版本已经修复

liaoya, 2025-06-23:
最新的代码里面问题又出现了。