destan19/OpenAppFilter #227: 编译失败：appfilter_netlink.c 中缺少 'hash_appid' 函数声明导致错误

lonecale, 2025-03-16:
在编译 luci-app-oaf 插件时，编译器在 appfilter_netlink.c 文件中提示 “implicit declaration of function 'hash_appid'” 错误，导致编译中断。

下面是编译的错误日志：
`2025-03-16T05:27:49.1765264Z make[3]: Entering directory '/builder/openwrt/package/new/luci-app-oaf/oaf'
2025-03-16T05:27:49.4608315Z mkdir -p /builder/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
2025-03-16T05:27:49.4634850Z SHELL= flock /builder/openwrt/tmp/.root-copy.flock -c 'cp -fpR /builder/openwrt/build_dir/target-x86_64_musl/linux-x86_64/oaf/.pkgdir/kmod-oaf/. /builder/openwrt/staging_dir/target-x86_64_musl/root-x86/'
2025-03-16T05:27:49.4691112Z touch /builder/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.kmod-oaf_installed
2025-03-16T05:27:49.4784323Z echo "kmod-oaf" >> /builder/openwrt/staging_dir/target-x86_64_musl/pkginfo/oaf.default.install
2025-03-16T05:27:49.4802871Z make[3]: Leaving directory '/builder/openwrt/package/new/luci-app-oaf/oaf'
2025-03-16T05:27:49.4808754Z time: package/new/luci-app-oaf/oaf/compile#0.19#0.13#0.31
2025-03-16T05:27:49.4930131Z make[3]: Entering directory '/builder/openwrt/package/new/luci-app-oaf/open-app-filter'
2025-03-16T05:27:49.7610118Z rm -f /builder/openwrt/build_dir/target-x86_64_musl/appfilter/.built
2025-03-16T05:27:49.7636580Z touch /builder/openwrt/build_dir/target-x86_64_musl/appfilter/.built_check
2025-03-16T05:27:49.7663150Z make -C /builder/openwrt/build_dir/target-x86_64_musl/appfilter/ CC="x86_64-openwrt-linux-musl-gcc" CFLAGS="-O3 -mtune=generic -pipe  -fhonour-copts -fmacro-prefix-map=/builder/openwrt/build_dir/target-x86_64_musl/appfilter=appfilter -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LIBS="-L/builder/openwrt/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/usr/lib -L/builder/openwrt/staging_dir/toolchain-x86_64_gcc-14.2.0_musl/lib -fuse-ld=mold -Wl,--gc-sections -flto=auto -fuse-linker-plugin -znow -zrelro -lm -lpthread -lubox -luci -lubus -ljson-c -lblobmsg_json" all
2025-03-16T05:27:49.7690947Z make[4]: Entering directory '/builder/openwrt/build_dir/target-x86_64_musl/appfilter'
2025-03-16T05:27:49.7692626Z x86_64-openwrt-linux-musl-gcc -O3 -mtune=generic -pipe  -fhonour-copts -fmacro-prefix-map=/builder/openwrt/build_dir/target-x86_64_musl/appfilter=appfilter -ffunction-sections -fdata-sections -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -c -o appfilter_netlink.o appfilter_netlink.c
2025-03-16T05:27:49.7911811Z appfilter_netlink.c: In function 'appfilter_nl_handler':
2025-03-16T05:27:49.7912889Z appfilter_netlink.c:106:9: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
2025-03-16T05:27:49.7913890Z   106 |     mac = json_object_get_string(mac_obj);
2025-03-16T05:27:49.7914324Z       |         ^
2025-03-16T05:27:49.7924150Z appfilter_netlink.c:153:20: error: implicit declaration of function 'hash_appid' [-Wimplicit-function-declaration]
2025-03-16T05:27:49.7925377Z   153 |         int hash = hash_appid(appid);
2025-03-16T05:27:49.7926024Z       |                    ^~~~~~~~~~
2025-03-16T05:27:49.7953394Z make[4]: *** [<builtin>: appfilter_netlink.o] Error 1
2025-03-16T05:27:49.7954116Z make[4]: Leaving directory '/builder/openwrt/build_dir/target-x86_64_musl/appfilter'
2025-03-16T05:27:49.7955035Z make[3]: *** [Makefile:59: /builder/openwrt/build_dir/target-x86_64_musl/appfilter/.built] Error 2
2025-03-16T05:27:49.7956048Z make[3]: Leaving directory '/builder/openwrt/package/new/luci-app-oaf/open-app-filter'
2025-03-16T05:27:49.7961241Z time: package/new/luci-app-oaf/open-app-filter/compile#0.19#0.13#0.31
2025-03-16T05:27:49.7968558Z     ERROR: package/new/luci-app-oaf/open-app-filter failed to build.
2025-03-16T05:27:49.7972350Z make[2]: *** [package/Makefile:129: package/new/luci-app-oaf/open-app-filter/compile] Error 1
2025-03-16T05:27:49.7984140Z make[2]: Leaving directory '/builder/openwrt'
2025-03-16T05:27:49.8016987Z make[1]: *** [package/Makefile:123: /builder/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2025-03-16T05:27:49.8028162Z make[1]: Leaving directory '/builder/openwrt'
2025-03-16T05:27:49.8056552Z make: *** [/builder/openwrt/include/toplevel.mk:233: world] Error 2
2025-03-16T05:27:49.8063355Z ##[error]Process completed with exit code 2.
`

ilxp, 2025-03-16:
和我的一样 https://github.com/destan19/OpenAppFilter/issues/226

destan19, 2025-03-18:
修复了编译错误

closed 2025-03-18 by destan19

lonecale, 2025-03-18:
https://github.com/destan19/OpenAppFilter/blob/72ff7c40cc166b3e468e1954742c51dac96930ed/oaf/src/af_log.c#L118

6.11 内核 也不要 oaf_table 数组中添加一个空的表项 {}
我现在是删除了这个判断 或者您这边能改下
#if (LINUX_VERSION_CODE < KERNEL_VERSION(6, 11, 0))