destan19/OpenAppFilter #265: 如何为普通的Linux kernel （如https://github.com/ophub/kernel/) 编译 oaf 内核模块

liaoya, 2025-07-25:
敬爱的作者：

[amlogic-s9xxx-openwrt](https://github.com/ophub/amlogic-s9xxx-openwrt) 支持很多 arm 的盒子，但是它使用了自己的内核而不是对应 openwrt 的。请问如何编译相应的内核模块，就可以支持更多的设备了。
谢谢！

GroverLau, 2025-08-29:
[amlogic-s9xxx-armbian](https://github.com/ophub/amlogic-s9xxx-armbian)提供了内核一键编译脚本,可以自行添加oaf驱动源码到内核目录编译.或者内核编译完成之后使用make -C 指定内核目录 M=oaf目录 modules 编译oaf驱动

GroverLau, 2025-09-02:
> > [amlogic-s9xxx-armbian](https://github.com/ophub/amlogic-s9xxx-armbian)提供了内核一键编译脚本，可以自己添加oaf驱动源码到内核目录编译。或者内核编译完成后进入oaf驱动源码目录，，使用make -C 指定内核目录 M=$(pwd) 模块编译驱动。注意make时需要配置目标平台架构以及工具链等环境变量
> 
> 老大有尝试编译出来没？我按照ophub提供的添加自定义驱动章节操作，未能成功make（执行make直接报错了），本仓库提供的源码不是不能直接在armbian系统里编译，要结合OpenWrt上编译才行（不是很懂Linux驱动这一块）

我在WSL2环境下实测可以编译，系统版本Ubuntu 20.04，目标平台rockchip

laiyujun, 2025-09-02:
最新进展：写了个Makefile编译成功并成功加载驱动

[image]

laiyujun, 2025-09-13:
最新进展：已编译进内核，可以使用该内核直接打包openwrt或armbian

[image]

liaoya, 2025-09-16:
> > 最新进展：已编译进内核，可以使用该内核直接打包openwrt或armbian
> > [image]
> 
> 1、可以使用我的仓库编译的内核自己打包固件，kernel_repo: https://github.com/laiyujun/kernel/releases 2、如果是N1的话也可以直接使用我打包好的固件，https://github.com/laiyujun/build-openwrt/releases

非常感谢，我尽快试试，不过我用的 firefly-rk3399 默认用的是 6.1 的 kernel，我看看是否可以切换到 6.6 的 kernel。

liaoya, 2025-10-04:
@laiyujun 
手工在晶晨宝盒里升级到 6.6 kernel，`oaf.ko` 被正确加载。<https://github.com/laiyujun/kernel/releases/download/kernel_stable/6.1.154.tar.gz> 并没有 `oaf.ko`。我看了 https://github.com/laiyujun/kernel/tree/main/kernel-patch/beta/linux-6.6.y 只 patch 了 6.6 内核。不知道能否 patch 6.1 内核。

destan19, 2026-11-11:
其他第三方的源码需要自行编译，ko模块严格依赖内核版本和配置，并不能通用

closed 2026-11-11 by destan19

laiyujun, 2026-11-12:
> [@laiyujun](https://github.com/laiyujun) 手工在晶晨宝盒里升级到 6.6 kernel，`oaf.ko` 被正确加载。非常感谢！！ https://github.com/laiyujun/kernel/releases/download/kernel_stable/6.1.154.tar.gz 并没有 `oaf.ko`。我看了 https://github.com/laiyujun/kernel/tree/main/kernel-patch/beta/linux-6.6.y 只 patch 了 6.6 内核。不知道能否 patch 6.1 内核。

该bug在你提出后就已经修复，现在编译的所有版本内核都带有oaf.ko